bigtreetech/BIGTREETECH-GTR-V1.0 #1: Schematics

Coffee0297, 2020-01-17:
we need schematic files for this board to be open source. please upload this.

TheMiguelBi, 2020-02-04:
...and the .STL file! ;)

G4Cab, 2020-03-22:
Please BigTreeTech, provide the schematics!!!

DJDOUBLEG, 2020-03-25:
done

Markus-1984, 2020-05-01:
Please provide the schematics of the M5, too.

danym21, 2020-05-16:
can we have also the STL files please?

GadgetAngel, 2020-06-01:
Please provide the schematics for the M5 board

GadgetAngel, 2020-06-03:
The BTT Schematic shows + sign on the HEATERS of the MOS DRIVERS block on top of the arrow which is wrong!  The + sign needs to be below the arrow.  Here is the corrected version (see the yellow highlight in the picture below):

[GTR V1.0(SCH)_Yellow.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4722726/GTR.V1.0.SCH._Yellow.pdf)

[image]

ddB0515, 2020-07-27:
@bigtreetech I'm not sure if anyone read this but there is NO M5 board schematic yet
I know you are busy abut would be nice to provide pdf for M5 board
Thanks in advance

G4Cab, 2020-07-27:
> @bigtreetech I'm not sure if anyone read this but there is NO M5 board schematic yet
> I know you are busy abut would be nice to provide pdf for M5 board
> Thanks in advance

There's an inofficial version of Thomas White in the facebook group

https://m.facebook.com/groups/485785992127996?view=permalink&id=566686720704589

bosd, 2020-11-26:
Here's the schematic, so facebook less users can access it.
[BIGTREETECH M5 V1.0-Schematics.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/5602206/BIGTREETECH.M5.V1.0-Schematics.pdf)