bigtreetech/BIGTREETECH-GTR-V1.0 #4: EEPROM not used?

Mosher23, 2020-02-15:
so i have seen that you have added onboard EEPROM, but i did't find anywhere in firmware use of it!
Please change it!

MangaValk, 2020-02-15:
I was able to enable it, but it crashes everytime when saving

Rincewind08, 2020-03-05:
i think i had found it in the pin file, but it lists several options for using eeprom.  i am using the sd card eeprom method and haven't had any trouble with this in particular.

Rincewind08, 2020-03-05:
I am starting to imagine that our friends at BIGTREETECH have been quarantined.

Rincewind08, 2020-03-05:
as described in pin file 

// Use one of these or SDCard-based Emulation will be used
//#define I2C_EEPROM
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
//#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation

alieneila, 2020-03-11:
I've tried all 3 settings and only the emulated ones work. Even the firmware BTT provides the EEPROM is disabled, so would love to see a response from them.

Rincewind08, 2020-03-12:
Likewise. Only the as card based eeprom will even compile. I believe our Chinese based friends are under house arrest more or less. They haven't put out hardly anything on GitHub, customer support is absent. I only hope the developers survive.

Rincewind08, 2020-03-12:
*sd based eeprom

Rincewind08, 2020-03-17:
I can verify successful printing with "flash" based eeprom, but if you change your sd card and reboot printer without it, expect trouble.  I am having a whole lot of problems getting consistent behavior out of my bl touch using tmc2209s, i cannot use them without the bltouch refusing to keep its pin up.  Very curious problems.  I have raised my steppers up on the 8 pin headers to raise the driver on both sides so that neither of the two extra pins touch their slots, and have still not made any progress.  For a moment it cooperated with 2209s, but it was a short moment, as one reboot later it refused to operate properly again.  I just wanted to use this board with all the features advertised.  I don't know if anyone from bigtreetech is left, but if so, could you please reach out to some of us with some real information?  Many of us have hundreds of dollars tied up in your products, and frankly, I'm starting to feel like a fool for purchasing this "top of the line" wonder board you folks produced.  I get better operation from a 1.3 or 1.4.

DJDOUBLEG, 2020-03-24:
> I was able to enable it, but it crashes everytime when saving

I have same issue.
Total clear Marlin 2.0.5.

https://youtu.be/4wXI1qSCrEE

dineshvyas, 2020-03-26:
When i was ordering GTR 1 I postponed at last moment. Since I had bad experience with BTT SKR PRO_1.1 super hyped board but nothing worked in actual. BIQU RUMBA is 100 times better.

DJDOUBLEG, 2020-03-26:
PRO working for me perfect!

DPrime <notifications@github.com> ezt írta (időpont: 2020. márc. 26., Cs
8:28):

berrenger, 2020-04-04:
I solved this problem.
use this link:
https://www.instructables.com/id/Correcting-the-Issue-With-On-board-EEPROM-for-the-/

[image]

NAPCAL, 2020-04-25:
> I solved this problem.
> use this link:
> https://www.instructables.com/id/Correcting-the-Issue-With-On-board-EEPROM-for-the-/
> 
> [image]

I think I solved it, that’s my instructable.

berrenger, 2020-04-25:
> > I solved this problem.
> > use this link:
> > https://www.instructables.com/id/Correcting-the-Issue-With-On-board-EEPROM-for-the-/
> > [image]
> 
> I think I solved it, that’s my instructable.

Sorry for my bad english. I mean my problem has solved.

NAPCAL, 2020-04-25:
> > > I solved this problem.
> > > use this link:
> > > https://www.instructables.com/id/Correcting-the-Issue-With-On-board-EEPROM-for-the-/
> > > [image]
> > 
> > 
> > I think I solved it, that’s my instructable.
> 
> Sorry for my bad english. I mean my problem has solved.

No problem, just picking on you, thanks for sharing my Instructable.

thisiskeithb, 2020-05-18:
I submitted a PR for native EEPROM support for the GTR: https://github.com/MarlinFirmware/Marlin/pull/18029