bigtreetech/BIGTREETECH-GTR-V1.0 #5: Pin file and manual figure do not match board silk screen

jonmstraub, 2020-02-21:
pins_BTT_GTR_V1_0.h and page 9, figure 9 of the manual have pin labels that do not match the silkscreen on the GTR v1.0 board.

The picture included in the repository, GTR V1.0-BOTTOM.png has the current silkscreen labels shown.

Which pin labels are correct?

bakaufman, 2020-03-01:
In the Marlin file pins_BTT_GTR_V1.0.h, the third driver pin definition (i.e. Z) calls for #define Z_DIR_PIN          PB7, but calls for PC7 in the GTR V1.0 Pin.PDF. Which is correct? Can anyone that has Z motion on that driver relay their pin settings please? Thanks!

Rincewind08, 2020-03-05:
I cannot get my z to go in the correct direction no matter what i invert my z to.  i think this might be related?

Rincewind08, 2020-03-05:
also, having tried in firmware and physically both the z min and the probe row z sensor i cannot in any configuration get my bltouch to actually trigger when pressed.  only when using z min as probe pin can i get it to deploy the pin as it is supposed to.  any insight into this would be greatly appreciated.

bakaufman, 2020-03-05:
how many Z motors are you running? I have 3 so it warrants asking.
I had to put the probe pins in the Z min position and not the extension directly below the probe. Confirm that you can run ABL test.

Rincewind08, 2020-03-06:
I am running two zed motors. I tried a hundred times to run it without doing their tandem z axis bullshit, but I ended up having to use only one zed driver, used a 2208 instead of a 2209, set it up to use the z min endstop as z min probe pin, and while UBL wasn't working right bilinear levelling works amazingly an very quickly. I have the issue with UBL that the probe will only deploy once per boot, after that it never deploys but rams into z axis or triggers early. With both the SKR 1.4 and the GTR v1.0 I have never had any success with the  BLTouch ports by the servo.

Rincewind08, 2020-03-06:
I think it had something to do with the DIAG feedback from the 2209s. It even with 2208s I cannot make this board home with two z drivers.  I will try with 2208 standalones and 2225s today. I am now using the Marlin 2.0 development from last night, the 2.1 development version doesn't have the GTR listed in the boards.h files despite having the environment. Also, the new Marlin 2.0 development version doesn't have any issues compiling on VSCODE on LinuxMINT.

bakaufman, 2020-03-06:
> I think it had something to do with the DIAG feedback from the 2209s. It even with 2208s I cannot make this board home with two z drivers. I will try with 2208 standalones and 2225s today. I am now [...]

Sounds complicated. Using TMC2130 in SPI and endstops plus BLtouch in probe + z endstop position, I can get XYZ homing (G28), z auto align (G34), and G28 P1/3 to work. 2.0.4.4. I did cut the diag pin, but could have just pulled the jumped down by that endstop pins.

NAPCAL, 2020-03-10:
https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/issues/12