bigtreetech/BIGTREETECH-GTR-V1.0 #6: MARLIN FIRMWARE NOT COMPILING

Rincewind08, 2020-02-24:
I cannot get the GTR firmware to compile using ATOM, or VSCODE, the guide that the documentation refers to is for another board entirely, I spent 70 bucks getting this thing to me and I'm very offset that the firmware hasn't seen more development since its release.

It throws more errors than I can even list.  Starts going on about pins being redfined left and right, void space will not start, pages and pages of errors.  Please give me something to work with here, I have essentially no information on how this board is meant to be coded.

JulianSimpsonWork, 2020-02-24:
I've been able to compile it fine, can you list the actual errors you get, dont forget one mistake can cause a knock on effect and make many things fail to compile.  Usually one error make it look like you have 20

JulianSimpsonWork, 2020-02-24:
It does sound like you dont have the board set in Platformio properly if its whinging about pins

Look in platformio.ini for this 

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_GTR_V1_0

Rincewind08, 2020-02-24:
What should it say in boards directory? I knew to set my environment in Marlin, not exactly my first rodeo, set up dozens of boards on Marlin, but never an arm and the only atom I've used is with the skr boards series. Please man enlighten me.

JulianSimpsonWork, 2020-02-24:
I download the basic repo from GitHub Bigtreetech, Github shows I changed 3 files to get it to compile for my setup, I added a _statusscreen.h and made changes to configuration and configuration_adv to reflect my setup.  Other than that everything is as it came from BigTreeTech, no point me showing you my boards file as its exactly as it came from GitHub.  If you provide the first few lines of errors its probably going to give a lot of us a real view of your problem.

JulianSimpsonWork, 2020-02-24:
correction _bootscreen.h not _statusscreen.h

> Executing task: C:\Users\Julian.Simpson\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_GTR_V1_0 (platform: ststm32@>=5.7.0; framework: arduino; board: BigTree_SKR_Pro)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.0.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 3.10700.191028 (1.7.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.3
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\Print.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\WString.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\abi.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\board.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\hooks.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\itoa.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\main.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\new.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
[... 834 lines cut ...]
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_GTR_V1_0                SUCCESS   00:08:49.864
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
===================================================================================== 1 succeeded in 00:08:49.864 =====================================================================================

Terminal will be reused by tasks, press any key to close it.

JulianSimpsonWork, 2020-02-24:
[image]

JulianSimpsonWork, 2020-02-24:
Make sure all the dependencies are met and versions are as above

tbojim1975, 2020-02-24:
[image]

My board is a Bigtreetech GTR v1.0 and stepper drivers  tmc2208 uart six drivers will be installing an on a ender3 (brand new from creality) with the original display. In vscode v1.42 with platformio installed. I continue to get this fail while compiling? Any ideas would be  Greatly appreciated.

JulianSimpsonWork, 2020-02-24:
Disable your LCD screen(s) from configuration.h lines 1564-2066 and try to get it to compile without screen support.  I'm using the original Ender3 screen myself and 5 TMC2208's so pretty similar setup stepper/screen wise

JulianSimpsonWork, 2020-02-24:
and this is how my screen is setup
[image]

tbojim1975, 2020-02-24:
I have a new issue?

tbojim1975, 2020-02-24:
I kept messing with it then got the first issue resolved by completely
starting from the beginning

tbojim1975, 2020-02-25:
I wanted to say thanks for helping me with the first issue. 
  Here's my new issue. 
I got one line to go away but I am stuck now? Still have one on line 63. Not sure where to put the value and which value. 
 [image]

JulianSimpsonWork, 2020-02-25:
check configuration.h for this section, serial ports are defined as -1 and 3

// @section machine

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3

Rincewind08, 2020-02-25:
> Executing task in folder Marlin-bugfix-2.0.x BTT_GTR_V1.0-Demo: C:\Users\Rwind\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_GTR_V1_0 (platform: ststm32@>=5.7.0; framework: arduino; board: BigTree_SKR_Pro)
--------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.0.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10800.200207 (1.8.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.4
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <IWatchdog> 1.0.0
|-- <Servo> 1.1.2
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduinoVariant\PeripheralPins.c.o  
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduinoVariant\variant.cpp.o       
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\HardwareSerial.cpp.o       
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\HardwareTimer.cpp.o        
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\Print.cpp.o
In file included from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring.h:41,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\Arduino.h:32,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.cpp:26:
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'      
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_DisableBRK(TIM_TypeDef*)':
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2814:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'      
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_ConfigBRK(TIM_TypeDef*, uint32_t)':
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'      
   (void)(tmpreg);
         ~^~~~~~~
In file included from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring.h:41,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\Arduino.h:32,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareTimer.cpp:26:
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'      
   (void)(tmpreg);
[... 151 lines cut ...]
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring.h:41,
[... cut ...]
C:\Users\Rwind\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'      
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\wiring_analog.c.o
   (void)(tmpreg);
         ~^~~~~~~Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\wiring_digital.c.o

Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\wiring_pulse.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\wiring_shift.c.o
Compiling .pio\build\BIGTREE_GTR_V1_0\FrameworkArduino\wiring_time.c.o
Compilin

Rincewind08, 2020-02-25:
this is driving me completely insane.  i've been coding marlin for years.  why doesn't this want to chooch?

Rincewind08, 2020-02-25:
This is with no changes other than drivers stated as tmc2130s.  Have also tried 2208s and 2209s, makes no difference. what am i missing man?

> Executing task in folder Marlin-bugfix-2.0.x BTT_GTR_V1.0-Demo: C:\Users\Rwind\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_GTR_V1_0 (platform: ststm32@>=5.7.0; framework: arduino; board: BigTree_SKR_Pro)
--------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.0.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10800.200207 (1.8.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.4
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <IWatchdog> 1.0.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
Building in release mode
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\HAL\HAL_STM32\HAL.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\HAL\HAL_STM32\HAL_SPI.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\HAL\HAL_STM32\Servo.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\HAL\HAL_STM32\SoftwareSerial.cpp.o
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32\timers.h:25,
                 from Marlin\src\HAL\HAL_STM32\HAL.h:99,
                 from Marlin\src\HAL\HAL_STM32\HAL.cpp:26:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1249:6: error: #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
     #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
      ^~~~~
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:2367:4: error: #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
   #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32\HAL_SPI.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1249:6: error: #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
     #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
      ^~~~~
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:2367:4: error: #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
   #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:38:0,     
                 from Marlin\src\HAL\HAL_STM32\Servo.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1249:6: error: #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
     #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
      ^~~~~
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:2367:4: error: #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
   #error "PRINTCOUNTER requires EEPROM_SETTINGS. Please update your Configuration."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32\timers.h:25,
                 from Marlin\src\HAL\HAL_STM32\SoftwareSerial.cpp:40:
[... 124 lines cut ...]
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_GTR_V1_0                FAILED    00:00:23.456
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
===================== 1 failed, 0 succeeded in 00:00:23.456 =====================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

JulianSimpsonWork, 2020-02-25:
you say you made no changes but you've turned on UBL bed levelling as its whinging about that, find eprom settings in the config file and turn it on but before that you need to start with the base repo and make sure that will compile before you change anything like stepper drivers and bed levelling.  My suggestion would be to start again, only making changes one by one and compiling in between to make sure your last change didnt break something.  Use GitHub desktop to track your personal changes to the repo and it will give you a nice list of all changes you have made

klucky, 2020-02-25:
I can't compile it too. Without any changes i get this:
                                                                                                          ^
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.h:225:71: warning: 'boolean' is deprecated [-Wdeprecated-declarations]
 void fio_shiftOut1(uint8_t pin, uint8_t value, boolean noLatch = false);
                                                                       ^
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.cpp:202:35: warning: 'boolean' is deprecated [-Wdeprecated-declarations]
                    boolean noLatch)
                                   ^
Compiling .pio\build\BIGTREE_GTR_V1_0\lib5c7\TMCStepper_ID5513\source\CHOPCONF.cpp.o
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.cpp:275:63: warning: 'boolean' is deprecated [-Wdeprecated-declarations]
 void fio_shiftOut1(uint8_t pin, uint8_t value, boolean noLatch)
                                                               ^
In file included from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:32,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LCD.cpp:54:
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_DisableBRK(TIM_TypeDef*)':
Compiling .pio\build\BIGTREE_GTR_V1_0\lib5c7\TMCStepper_ID5513\source\COOLCONF.cpp.o
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2814:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_ConfigBRK(TIM_TypeDef*, uint32_t)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
In file included from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:32,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal.cpp:48:
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_DisableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2814:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_ConfigBRK(TIM_TypeDef*, uint32_t)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
[... cut ...]
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
=============================================================================== 1 failed, 0 succeeded in 00:00:11.832 ===============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
![error](https://user-images.githubusercontent.com/30538085/75285226-87421080-5816-11ea-9df3-e32a151a8334.png)

klucky, 2020-02-25:
[image]

tbojim1975, 2020-02-25:
Klucky looks like you are going to run in to another problem after you get the first one fixed.  I just had that issue and can not figure it out because I show no "warnings or problems" with this one. The "firmware.elf error" if you figure it out let me know. 
"Building in release mode
Linking .pio\build\BIGTREE_GTR_V1_0\firmware.elf
c:/users/klucky/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_GTR_V1_0\firmware.elf]"

nsmdigitalpc, 2020-02-26:
Hi JulianSimpsonWork, if your settings are the same as mine and yours worked . So i can be on my way with this board working, can you send me a copy of your marlin settings files folder in a zip because seems i keep hitting dead ends here. I'm losing a lot of sleep trying to get this board to work.

tbojim1975, 2020-02-26:
This is my last error I hope. 
[image]
Not sure

tbojim1975, 2020-02-26:
nsmdigitalpc is also me I did not realize my old account is still active. Anyway.....

JulianSimpsonWork, 2020-02-26:
[BTT-GTR-Marlin config for Marlin-bugfix-2.0.x BTT_GTR_V1.0-Demo.zip](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4257154/BTT-GTR-Marlin.config.for.Marlin-bugfix-2.0.x.BTT_GTR_V1.0-Demo.zip)

Note I use 3 extruders, BL touch, etc so you should not assume this will work for your printer without changes for your setup but it should compile OK

klucky, 2020-02-26:
I have now downloaded the firmware again and only repaced the config files from your zip file.
That is the result i get:

         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_ConfigBRK(TIM_TypeDef*, uint32_t)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
In file included from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:32,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal_SR.cpp:100:
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
In file included from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal.h:45:0,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal.cpp:50:
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.h: At global scope:
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.h:217:106: warning: 'boolean' is deprecated [-Wdeprecated-declarations]
 void fio_shiftOut1(fio_register shift1Register, fio_bit shift1Bit, uint8_t value, boolean noLatch = false);
                                                                                                          ^
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\FastIO.h:225:71: warning: 'boolean' is deprecated [-Wdeprecated-declarations]
 void fio_shiftOut1(uint8_t pin, uint8_t value, boolean noLatch = false);
                                                                       ^
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_DisableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2814:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_ConfigBRK(TIM_TypeDef*, uint32_t)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2834:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
In file included from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:32,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LCD.h:55,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal_SR2W.h:137,
                 from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal_ID136\LiquidCrystal_SR2W.cpp:44:
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':
C:\Users\klucky\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_tim.h:2797:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
   (void)(tmpreg);
         ~^~~~~~~
In file included from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL/stm32yyxx_ll_tim.h:21:0,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:32,
                 from C:\Users\klucky\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
[... cut ...]
=============================================================================== 1 failed, 0 succeeded in 00:00:11.924 ===============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Maybe there's something wrong with the platformio.ini?
I have even reinstalled vscode and pio.
I have even installed vscode as system installer and user installer but there was no difference.

klucky, 2020-02-26:
[image]

Which version of vscode do you use?
I'm using:
Version: 1.42.1 (user setup)
Commit: c47d83b293181d9be64f27ff093689e8e7aed054
Date: 2020-02-11T14:45:59.656Z
Electron: 6.1.6
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.18363

JulianSimpsonWork, 2020-02-26:
> CAUTION: This email originated from outside of the organisation. Do not click links or open attachments unless you recognise the sender and know the content is safe.
>
> [vscode1]<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F30538085%2F75390311-58946a80-58e8-11ea-8619-927a686c6f40.png&data=02%7C01%7CJulian.Simpson%40emishealth.com%7C0b695cee1aec493a10f608d7bb040770%7C982035bb0c384f2e8ce049e49bd4ee04%7C0%7C0%7C637183498289741577&sdata=ua28gdMg%2Fb8mWwNEr8xoaaBwWFehBV1np1z1a4L8kgI%3D&reserved=0>

Version: 1.42.0 (system setup)
Commit: ae08d5460b5a45169385ff3fd44208f431992451
Date: 2020-02-06T10:51:34.058Z
Electron: 6.1.6
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.18362

I don’t think it’s that, your dependencies look good too, I am one version behind on NeoPixel as I butchered mine to do a fancy start-up sequence so didn’t update it.  This is more of a linker error as it does compile.  Not sure exactly why at moment but its saying it cant find stm32f407xg.ld so have you checked to see if it exists?  If not look into why not and if it does exist why can’t the linker get to it.  Permissions I don’t think as you’ve tried admin users but I would check access on the PIO folder structure.

I think you may get a good response posting this on the main Marlin group as I don’t think this is GTR config related and there will be more views on there.

Linking .pio\build\BIGTREE_GTR_V1_0\firmware.elf
c:/users/klucky/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_GTR_V1_0\firmware.elf] Error 1

Julian

EMIS Health is a trading name used by members of the EMIS Group of companies which includes:
- Egton Medical Information Systems Limited. Registered number: 02117205.
- Ascribe Limited. Registered number: 02394847.
- Dovetail Digital Limited. Registered number: 10867961.
- Rx Systems Limited. Registered number: 04704728 (also trading as EMIS Health Community Pharmacy).
Each company listed above is registered in England and it's registered office address is: Fulford Grange, Micklefield Lane, Rawdon, Leeds LS19 6BA.
The information contained in this email (including any attachments) is confidential and may be legally privileged. It is solely for use by the intended recipient(s). If you are not an intended recipient you must not use or publish its contents. If you receive this email in error, please contact us immediately and permanently delete the email. Service can only be effected on us by email in accordance with unequivocal contractual provisions. Emails are not secure and may contain viruses. We may monitor traffic data. Further information is available at www.emishealth.com (including information about how we process data, as detailed in our Privacy policy).

nsmdigitalpc, 2020-02-27:
Thank you for that info Julian. I fixed the issue last night be fore i went to bed, final got it to compile. 
I updated all my lcd library.
then found a character that should not have been added as far as i can tell. Right after "=" and before the 3.10700.19102 it had a ">" I deleted this, then everything compiled and seems fine.

Bad -----> platform_packages = framework-arduinoststm32@=>3.10700.191028<-------  

Good -----> platform_packages = framework-arduinoststm32@=3.10700.191028<-------  
lol
the way mine looks now:
  # Bigtreetech GTR V1.0 (STM32F407IGT6 ARM Cortex-M4)
#
[env:BIGTREE_GTR_V1_0]
platform          = ststm32@>=5.7.0
framework         = arduino
platform_packages = framework-arduinoststm32@=3.10700.191028
board             = BigTree_SKR_Pro
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags       = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407IG\"
  -DTARGET_STM32F4 -DSTM32F407IX -DVECT_TAB_OFFSET=0x8000
  -DHAVE_HWSERIAL3
  -DHAVE_HWSERIAL6
  -DPIN_SERIAL3_RX=PD_9
  -DPIN_SERIAL3_TX=PD_8
  -DPIN_SERIAL6_RX=PC_7
  -DPIN_SERIAL6_TX=PC_6

  -IMarlin/src/HAL/HAL_STM32
lib_deps          =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal
  TMCStepper@>=0.5.2,<1.0.0
  Adafruit NeoPixel
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
lib_ignore        = SoftwareSerial, SoftwareSerialM
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed     = 250000

One last thing Julian do i need anything external, extra other then this board. example : can I plug the hot bed or even  hotend straight to the board without any chip between and everything else to the board as well? My goal for this board and printer to be a multi-color. i just want to make sure i have all my equipment i need. maybe even provide me with a quick list, wire diagram to measure up how i set mine up
ender 3 
printer display: KINGPRINT MKS TFT28 2.8-Inch
Board BTT GTR 1
setup will be Uart mode
tmc2208 with uart for 3 extruders eventually more
thc-01 hotend
6 Usongshine Us-17HS4401S STEPPER motor

JulianSimpsonWork, 2020-02-27:
You can run your hotend directly from the board and a DC hotbed but I'm working on an AC bed and using a couple of solid state relays to turn them on/off, my bed is using 2x400x400 1400W 240V AC's into a 6mm aluminium bed.   I have a massive issue in that I cant figure out how to control both beds with marlin, setting it up pin wise isnt the issue but the implications of trying to control 2 beds in parallel look like a rabbit hole.  Marlin devs appear to be running away from it so far so may have to do some research myself.

Rincewind08, 2020-02-27:
That is exactly what keeps happening to me. I have literally reinstalled Windows over it. I think it has something to do with symlinks, as if the program is missing components because it can't find the resources. For those who have been successful, how EXACTLY did you set up platformIo/vscode/linter?

klucky, 2020-02-27:
Thank you JulianSimpsonWork for your help!
And Thank you nsmdigitalpc for sharing your solution!
It was the same problem here and with this small change it compiled.
[image]
Now I am still getting some problems but it compiles anyway so i don't think this will matter.
[image]

nsmdigitalpc, 2020-02-28:
You are welcome. Yeah i still got that but not a big deal i believe. Let me
know your progress if everything works ok

bakaufman, 2020-02-28:
I removed the > as indicated above and it worked. A word of warning, in the .ini file, // isn't going to block the text, so if you leave yourself notes as I do use a # before the note.

bakaufman, 2020-02-28:
Note, someone on the GTR boards said to comment this out: 
#-DHAVE_HWSERIAL3 #jeff Org: Uncommented
  #-DHAVE_HWSERIAL6 #jeff Org: Uncommented
  #-DPIN_SERIAL3_RX=PD_9 #jeff Org: Uncommented
  #-DPIN_SERIAL3_TX=PD_8 #jeff Org: Uncommented
  #-DPIN_SERIAL6_RX=PC_7 #jeff Org: Uncommented
  #-DPIN_SERIAL6_TX=PC_6 #jeff Org: Uncommented 

Please let me know if the helps or hurts!

nsmdigitalpc, 2020-02-29:
Hey Julian I thought i was home free i'm having some issues now with my endstop not working with pronterface? Any issues with yours on the ender 3 factory endstops? Or what is the wire diagram if you know it off hand?

Rincewind08, 2020-03-01:
I was able to compile by installing linux mint (an ubuntu spinoff), setting up vscode and platformio, and cloning the directory.  i had to modify the Ini as nsmdigitalpc had suggested and everything compiled successfully without so much as a single warning in the compile.  it went rather beautifully.  if anyone is having problems compiling still, you can contact me at morelandmod@gmail.com and I will compile a version for you, assuming you supply the parameters for the printers equipment.  Issue closed.

closed 2020-03-01 by Rincewind08

JulianSimpsonWork, 2020-03-02:
> Hey Julian I thought i was home free i'm having some issues now with my endstop not working with pronterface? Any issues with yours on the ender 3 factory endstops? Or what is the wire diagram if you know it off hand?

I didnt use the factory endstops :)

tbojim1975, 2020-03-04:
Oh wow lol ! :-)  Pronterface is now reading my micro stops (open to trigger) but for some reason my end stops will not stop my x y z axis? I'm sure its a simple fix. If you have any ideas please let me know