bigtreetech/BIGTREETECH-GTR-V1.0 #8: Pinout figure in manual has an extra row of pins for the Raspberry Pi header connection.

jonmstraub, 2020-03-02:
Manual shows 21 rows of 2 pins for the Raspberry Pi header on the GTR board. The Raspberry Pi has a 40 pin (20 rows) header.

NAPCAL, 2020-03-10:
Here are
[GTR V1.0 PIn color AI.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4310239/GTR.V1.0.PIn.color.AI.pdf)
 the corrected pins.

GadgetAngel, 2020-05-31:
See issue "On pinout image - Z stepper DIR pin wrong #13".

Here are files with the correct PIN numbers (https://github.com/GadgetAngel/GTR-V1.0-Stepper-Driver-Jumper-Configuration-Manual): 

[GTR V1.0 Uncolored_PIn_Diagram_Final_300ppi.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4708319/GTR.V1.0.Uncolored_PIn_Diagram_Final_300ppi.pdf)
[GTR_V1.0_Colored_PIN_Diagram_FINAL_300ppi.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4708323/GTR_V1.0_Colored_PIN_Diagram_FINAL_300ppi.pdf)

[image]
[image]