bigtreetech/BIGTREETECH-GTR-V1.0 #9: Raspberry Pi will not boot when connected to GTR board, and power is shared between devices

jonmstraub, 2020-03-02:
The 5.0V and 3.3V power pins in the Raspberry Pi header should likely not be allowed into the 5V and 3.3V busses on the GTR board.

Each board will be overdrawn on current causing a low power condition for the Raspberry Pi. There should be some isolation here.

bakaufman, 2020-03-05:
Did you flip the ribbon cable and try?

clowrey, 2020-03-15:
Yes this is correct.. Only connect TX, RX, GND between them for a safe connection... design is crazy to have a 40 pin header when only 3 pins are needed.

DJDOUBLEG, 2020-04-27:
Cut the 3.3V pin!
If i use RPi2 then this pin not problem.
Now is use RPi 3B+ then not boot up when 3.3V is connected.

jonmstraub, 2020-05-08:
> Cut the 3.3V pin!
> If i use RPi2 then this pin not problem.
> Now is use RPi 3B+ then not boot up when 3.3V is connected.

Thank you for this info, though I understand there are workarounds. The point of opening the issue was to give feedback to the manufacturer to correct the problem in future releases.

Geraldblin, 2020-06-21:
Hi can someone help me configure this board thx