bigtreetech/BIGTREETECH-GTR-V1.0 #11: Klipper

Aloafromhell, 2020-03-08:
Hi guys. Who managed to configure Klipper with the gtr. I just got stuck had a config ready. However, no connection is made.

mak0t0san, 2020-07-09:
I did.  Which serial port are you trying to use? I connected my Raspberry Pi to the PI header pins on the GTR and using that serial port.

My mcu settings look like this:

```
[mcu]
baud: 250000
restart_method: command
serial: dev/serial0
```

shawnp30, 2020-10-19:
I’m also having trouble. I’m not getting anything from serial when using gnd rx and tx. Do I need to set parameters in make menuconfig. I disabled usb and went with serial uart1 with gpio pins enabled. Is this wrong? How can I fix it. ?