bigtreetech/BIGTREETECH-GTR-V1.0 #12: GTR Board routing issue for the Stallguard jumpers, M5 has is correct.

NAPCAL, 2020-03-10:
The GTR board for all six Stallguard jumpers pin two is not connected to any other part on the board. Pin 1 of the jumpers have the ARM input for that stop, and the driver diag pin connected on the same pin. See this pdf for the current schematic of the GTR end stops. 
[GTR End Stop Circuits.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4310257/GTR.End.Stop.Circuits.pdf)

The M5 board has them setup correctly.  The Stallguard jumpers pin one goes to just the driver diag pin, Stallguard jumpers pin two goes to the ARM input for that stop and to the end stop's circuit. See this pdf for the current schematic of the M5 end stops.
[M5 End Stop circuits.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4310270/M5.End.Stop.circuits.pdf)

Rincewind08, 2020-03-10:
Thank God for you. I knew I should have waited for version 1.1 before purchase.

bigtreetech, 2020-03-12:
The DIAG pin need to be cut off Only when TMC2209 is used and stallguard detection is not used.  and No change required in other cases.

mawa85, 2020-03-12:
wow you integrate a pin for this and do not wire it? thats crap. 
I won't cut a pin on a reusable tmc2209 because you deliver faulty products.

Rincewind08, 2020-03-12:
It can always just be resoldered, but I agree. And just a heads up to everyone, the Eryone drivers are no different with this board, neither are the FYSETC. I will test the MKS and Watterott tonight after work.

Rincewind08, 2020-05-31:
So, here's the dealio.  This board isn't getting any support, despite the fact the BTT continues to roll out products.  In January they were explosive about this god damn board coming out, and now that a few.... hundred?  thousand?  of us have wasted our money on it, they have more or less abandoned us completely with a half finished firmware they can't even keep up to date. The features promised with this board simply are not there, and when it comes to something as simple as endstops, I just can't understand how they could mess this up so terribly.  They have neither admitted to it nor attempted to contact those who purchased the boards from them in order to resolve these enormous problems.  

There are no 2209s which you can get the use of Stallguard4/Coolstep out of for the zed axis.  It is absolutely impossible.  I have never gotten the extra pins over on the side for the BLTOUCH endstop ground and signal to work either, and I'm at least adequate at writing marlin configurations.  I was excited beyond belief when I bought this board, and all we got was a world threatening disease and 90$ worth of board that at best will one day run well on a community produced firmware.  That firmware will have to be catered to this board, or risky hardware fixes will have to be attempted, in order to get the functionality of it.  I wish I had never bought this thing.  I have an SKR 1.3 and a 1.4 running great, have been since I bought them.  I'm afraid to leave this board printing when I'm not at home.  I cannot even get it to use half the options available for it's EEPROM described in it's own pins file.

If anyone wants to get into this with me, I've not been able to find a working branch that can handle simple I2C Encoders, so I'm going to have to start with whats left of Marlin (as of the last week, the development branch disappeared entirely) or perhaps RepRapFirmware or Teacup and attempt to work out these issues one by one.  Would love some company.  MorelandMod@gmail.com.

NAPCAL, 2020-05-31:
> So, here's the dealio. This board isn't getting any support, despite the fact the BTT continues to roll out products. In January they were explosive about this god damn board coming out, and now that a few.... hundred? thousand? of us have wasted our money on it, they have more or less abandoned us completely with a half finished firmware they can't even keep up to date. The features promised with this board simply are not there, and when it comes to something as simple as endstops, I just can't understand how they could mess this up so terribly. They have neither admitted to it nor attempted to contact those who purchased the boards from them in order to resolve these enormous problems.
> 
> There are no 2209s which you can get the use of Stallguard4/Coolstep out of for the zed axis. It is absolutely impossible. I have never gotten the extra pins over on the side for the BLTOUCH endstop ground and signal to work either, and I'm at least adequate at writing marlin configurations. I was excited beyond belief when I bought this board, and all we got was a world threatening disease and 90$ worth of board that at best will one day run well on a community produced firmware. That firmware will have to be catered to this board, or risky hardware fixes will have to be attempted, in order to get the functionality of it. I wish I had never bought this thing. I have an SKR 1.3 and a 1.4 running great, have been since I bought them. I'm afraid to leave this board printing when I'm not at home. I cannot even get it to use half the options available for it's EEPROM described in it's own pins file.
> 
> If anyone wants to get into this with me, I've not been able to find a working branch that can handle simple I2C Encoders, so I'm going to have to start with whats left of Marlin (as of the last week, the development branch disappeared entirely) or perhaps RepRapFirmware or Teacup and attempt to work out these issues one by one. I would love some company. [MorelandMod@gmail.com](mailto:MorelandMod@gmail.com).

Sorry to hear your issues, but anyone can make a mistake. As for the firmware, BTT or any of their companies creates Marlin firmware Marlin group does, that part of your issue needs to be on Marlin’s GitHub. On the EEPROM issue, I found the temporary fix that BTT did incorporate into their copy of Marlin for the GTR. Marlin group did come up with an official fix for the EEPROM issue in their current Bugfix branch; it has been tested and works.

closed 2020-05-31 by NAPCAL