bigtreetech/BIGTREETECH-GTR-V1.0 #13: On pinout image - Z stepper DIR pin wrong

clowrey, 2020-03-15:
The Z stepper DIR pin is PB7 not PC7

https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/blob/master/BTT%20GTR%20V1.0/manual/GTR%20V1.0%20PIn.PDF

Can be verified - the Marlin PINS file has the correct PB7 pin defiined.

JaredC01, 2020-05-09:
Just had this same issue tonight.  After going to the schematic from the pinout diagram, I found the issue was the same as this currently has.  Can you PLEASE update this?!?

GadgetAngel, 2020-05-31:
There are really **5** items that are "wrong or missing or added that should not be" to the original BTT PIN Diagram.  
**#1** Z DIR PIN is PC7 **BUT** Marlin 2.0.5.x's pins_BTT_GTR_V1_0.h file shows it to PB7.
**#2** EXTENSION-1 Header PIN number 2 is PH10 **BUT** the BTT's schematic shows it to be PI10.
**#3** Raspberry pi Header Column 1 and Column 2 both have a count of 21 PINS, **BUT** BTT's schematic only shows 20 PINS in each Column.  There is one extra "NC" PIN in each column. See my attached file for which two PINS **"were added that should NOT be there"**.
**#4** The Stallguard BLOCK **is MISSING** the PIN number or name for the PIN that each DIAG1 PIN is connected up with.   **BUT**, the BTT's schematic indicates that the DIAG1 PIN from each Motor Drive is connected to its corresponding Limit Switch/Endstop. So X-DIAG1 should have X- (PF2) added; Y-DIAG1 should have Y- (PC13) added; Z-DIAG1 should have Z- (PE0) added; E0-DIAG1 should have E0 (PG14) added; E1-DIAG1 should have E1 (PG9) added; and E2-DIAG1 should have E2 (PD3) added.
**#5** The HOT BED or BED OUT **is MISSING** a PIN while the Marlin 2.0.5.x's pins_BTT_GTR_V1_0.h file shows it should be PA2, which is missing so it needs to be added to the PIN diagram.

I have created a PIN file showing these correction and a Color_PIN file also showing these corrections.  They both can be found at my github page: https://github.com/GadgetAngel/GTR-V1.0-Stepper-Driver-Jumper-Configuration-Manual

Here are the attached documents: 
[GTR V1.0 Uncolored_PIn_Diagram_Final_300ppi.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4708250/GTR.V1.0.Uncolored_PIn_Diagram_Final_300ppi.pdf)
[GTR_V1.0_Colored_PIN_Diagram_FINAL_300ppi.pdf](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4708251/GTR_V1.0_Colored_PIN_Diagram_FINAL_300ppi.pdf)
[image]
[image]

DrumClock, 2020-11-01:
Hi @GadgetAngel 

> I have created a PIN file showing these correction and a Color_PIN file also showing these corrections. They both can be found at my github page: https://github.com/GadgetAngel/GTR-V1.0-Stepper-Driver-Jumper-Configuration-Manual

**after unpacking the archive, the files have only 1kB and are damaged**

A13xund3r, 2024-03-30:
4 years later still not fixed... what a shame...