bigtreetech/BIGTREETECH-GTR-V1.0 #16: PROBLEM WITH MARLIN FIRMWARE

Symbyos83, 2020-04-12:
Hi, I have noticed that I have many problems with this card. I have loaded the marlin firmware with the deafult settings (serial ports, motor drivers, etc ...) but I cannot communicate the card with the PC or with the serial TFT35 display. Could anyone give me some tips? now I'm feeding it via USB of the pc.
Thank you.

nsmdigitalpc, 2020-04-12:
We need more details: what board, what tft35 display, the printer you are using and so on.......
1.check cables, make sure you have the correct connections
2.check the firmware on the controller board.
3.check the display tft 3.5 firmware (separate from controller board) and so on.
If you don't give details, trying to help you will take forever....

nsmdigitalpc, 2020-04-12:
* :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

/**
 * This setting determines the communication speed of the printer.

nsmdigitalpc, 2020-04-12:
I believe my gtr v1.0 were set like this 
* :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3

/**
 * This setting determines the communication speed of the printer.

TheMiguelBi, 2020-04-12:
I can answer for Symbyos...

The card is a:   GTR v1.0 
Display:            3.5 "TFT display E3 version. 
Backup:            24v UPS 
Drivers:             4 Drivers (X, Y, Z, E0) TMC2209 in UART mode.
Neopixel (16) and BLtouch.
No other extras

And yes, the serial connections have been set to -1 and 3.
That said, the display continues to show: "Printer not connected".

TheMiguelBi, 2020-04-12:
The Display is equipped with firmware: BIGTREE_TFT35_V3.0_E3.26.bin (last available) and the board currently has Marlin Firmware bugfix-2.0.x downloaded directly from bigtreetech github.

TheMiguelBi, 2020-04-14:
Ok. The problem has been discussed and resolved here: [https://github.com/MarlinFirmware/Marlin/issues/17297]
Symbyos83 you can close! 👍

closed 2020-04-14 by Symbyos83