bigtreetech/BIGTREETECH-GTR-V1.0 #17: Neopixel not function

Symbyos83, 2020-04-18:
Hi, I connected 16 neopixel leds (2 strips of 8 leds) to the 5V, PF13 and GND pins related to the neopixel on my GTR Pro v1.0. During the power-up phase, the self-test works correctly but when I start a print, they do not perform the color variation like the marlin configuration. could it be that the power pins are not enough to power 16 LEDs? why would they put the relevant pins?
Thank you.

TheMiguelBi, 2020-04-18:
UPDATE: by turning these 16 LEDs on manually via the TFT display, the LEDs light up in the desired color (remain lit even during printing).
A 5V power supply problem can therefore be ruled out ... 🤔

TheMiguelBi, 2020-04-18:
### 🚨🚨🚨
Maybe something is better explained here: https://github.com/MarlinFirmware/Marlin/issues/17024#issuecomment-595349265
### 🚨🚨🚨

TheMiguelBi, 2020-05-01:
I believe we can close this problem! 
The neopixel work regularly by initiating prints from the onboard microSD. The problems are caused by using the Graphic display **BTT_TFT35**! it's a real disaster!! the neopixel does not work during a print, the UPS does not work .....
There is a need to fix the Display firmware to solve this (and other) problems.

closed 2020-05-14 by Symbyos83