bigtreetech/BIGTREETECH-GTR-V1.0 #18: Rpi 3B+ and GTR power issue

DJDOUBLEG, 2020-04-27:
When i use RPi 2 then have no problem with internal power supply. Now i use RPi 3B+ then 5V/3A power is not enough (or the voltage is not enough).
Now i must use external power for this RPi.  **5.3V**(!!!)/5A.
The minimum voltage to RPi pin is 5.3V (if i set to 5.2 then always reboot or show octoprint undervoltage)
[image]

andysport, 2020-05-05:
Hi, If I understand correctly, you want power supply the RPi3 with GPIO port, right?
But I read in the raspberry support forum that RPi3 and RPi4 are not powered through the GPIO port.

DJDOUBLEG, 2020-05-05:
> Hi, If I understand correctly, you want power supply the RPi3 with GPIO port, right?
> But I read in the raspberry support forum that RPi3 and RPi4 are not powered through the GPIO port.

Ok. 
1. older RPi (like 1 or 2) not enough hw resource to print perfect curve without lag (stuttering)
2. if use  newest (like 3 or 4) not possible to powering over GPIO pin

Then why RPI GPIO interface was made for GTR. This is the question.

GadgetAngel, 2020-05-31:
> > Hi, If I understand correctly, you want power supply the RPi3 with GPIO port, right?
> > But I read in the raspberry support forum that RPi3 and RPi4 are not powered through the GPIO port.
> 
> Ok, then I have two remark:
> 
> 1. older RPi (like 1 or 2) not enough hw resource to print perfect curve without lag (stuttering)
> 2. if use newest (like 3 or 4) not possible to powering over GPIO pin
> 
> Then why RPI GPIO interface was made for GTR?
> This is the question.

the purpose it to get rid of the USB cable between the raspberry pi and the GTR USB port. By hooking up the raspberry pi thru the raspberry pi connector you no longer need a USB cable.

I am working on a wiring diagram now and will display it here when I get it done.