bigtreetech/BIGTREETECH-GTR-V1.0 #19: 12V fan on 24V supply

cscott, 2020-05-01:
24V fans are pretty hard to find. On the next revision of the hardware it might be nice to include a jumper to select either the 12/24V rail or the regulated 12V supply for the fan outputs.

NAPCAL, 2020-05-31:
If you have the M5 board it has that capability.

GadgetAngel, 2020-05-31:
> 24V fans are pretty hard to find. On the next revision of the hardware it might be nice to include a jumper to select either the 12/24V rail or the regulated 12V supply for the fan outputs.

Here is where I have found some:

https://www.amazon.com/gp/product/B07WWJF4LG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B0084ZNAU0/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B07L2YH79Y/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B07DB7DLMM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B07DB5H6K1/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

TH3D sells them
E3D online sells them

Exellent1988, 2020-11-30:
You could just change your max fan pwm to 120 in the config_advanced. So you'll get 12V on 100% :)

(please measure with an multimeter to get your correct max pwm value)

microtronics, 2021-08-31:
PWM and fans don't work that way. You'll still give pulses of 24V to your fans and depending on the fan it will take damage. I agree, a jumper solution for 24, 12 and 5 V is the way to go here