bigtreetech/BIGTREETECH-GTR-V1.0 #21: 5160 with GTR - timer problems solved?

Charly333, 2020-05-20:
Using 5160 with SKR 1.3 and SKR 1.1 Pro I had to cut timer pin and solder timer to ground to avoid timer overrun errors. Is this problem solved with GTR and M5 boards - or do I need to do the old workaround?

Rincewind08, 2020-05-31:
I have had no such problems.  only make sure that if you want to use your sensorless homing with the 5160s, including the temperature coolconf and the automatic amperage adjustment, that you utilize the diag pin, which is two over from the Enable pin in the little row where you would also put pins in for the encoders

GadgetAngel, 2020-05-31:
> I have had no such problems. only make sure that if you want to use your sensorless homing with the 5160s, including the temperature coolconf and the automatic amperage adjustment, that you utilize [...]

[image]
So you mean the DIAG1 PIN, correct?  Which pins need to be in for the encoders? Are ENCA and ENCB for encoders? Can you point to a source of information on how to use these pins?

Rincewind08, 2020-06-01:
All but the one diag pin are for an incremental encoder. Just solder on
your diag pin as illustrated on the card that came with the driver. If
someone has a working set of Marlin files, PLEASE prove me wrong and send
them on over.

Rincewind08, 2020-06-01:
I'm quite frustrated at the zed probe. Considering migrating the Cartesian movement to m5 board.