bigtreetech/BIGTREETECH-GTR-V1.0 #22: platformio.ini/env:/board = wrong board "BigTree_SKR_Pro"

b3nnyb3n, 2020-06-05:
{Hi! Thank-you to everyone that contributes!! }

[image]

after correcting env:"GTR"  /  board  to the correct board does not compile

if left unchanged compiles can be uploaded but: stepper motors not working 

work around so far... (*havent verified working yet)

left .ini file with error - specifying skr pro as - board  = BigTree_SKR_Pro

copied all text in the "pins_BTT_GTR_V1_0.h" file and pasted all to "pins_BTT_SKR_PRO_V1_1.h"

[marlin2.zip](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/files/4738786/marlin2.zip)
have included platformio.ini, .bin, and marlin folder

ucflumm, 2020-08-10:
getting the same problem here does pasting in the pins work?

ucflumm, 2020-08-10:
[image]

GadgetAngel, 2020-09-21:
I just compile my GTR V1.0 board using the following in platformio.ini file.  Ensure your [platformio] section looks as follows:
```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_GTR_V1_0
include_dir  = Marlin
```
[image]