bigtreetech/BIGTREETECH-GTR-V1.0 #24: Multiple Servos

codycato, 2020-06-10:
Nothing happens whenever i use Zmax or M5 Stop pins as servos. Only Bltouch works. But i need at least bltouch and another servo working.

I redefined pins and change num_servo to 2. no luck

geozoc, 2020-09-19:
Same problem here, I also need a second servo PIN. Which one to use?

matterhacker, 2020-11-03:
Trying to add another servo too, single servo works but only on PB11. Servo connected to other pins don't work when M280 command is used. Have tried with V2.0.x and V2.0.6.1 without success. Am I missing some defines somewhere? 
## Configuration.h
```c
#define NUM_SERVOS 2 // Servo index starts with 0 for M280 command

#define SERVO_DELAY { 300,300 }
```
Marlin\src\pins\stm32f4\pins_BTT_GTR_V1_0.h
```c
#if ENABLED(TP)
  #define SERVO0_PIN                        PB11  // original
  #define SERVO1_PIN                        PI7   // doesn't work with M280 P1 S128
#endif
```

goldensgui, 2020-11-21:
> Trying to add another servo too, single servo works but only on PB11. Servo connected to other pins don't work when M280 command is used. Have tried with V2.0.x and V2.0.6.1 without success. Am I [...]

Hello guys,

I've just tried with PH13 pins from the CAN Extension-0 (as I intend to use the M5 board, I did'nt want to use its dedicated pins) and it works without problem.

matterhacker, 2020-11-28:
Thanks @goldensgui I tried it out and can get PH13 going too !