bigtreetech/BIGTREETECH-GTR-V1.0 #30: USING ALL 6 DRIVERS

Gordon44, 2020-07-30:
Could someone answer my question?
I have the BIGTREETECH-GTR V1 and need to configure it using all 6 drivers one of them as extrude E0 and the other 5 drivers configured as drivers only. I am using the TMC2209 and can’t seem to get past 4 drivers X.Y,Z and E0 but can't get the other E1, E2 to operate the other two steppers 
How do I tell Marlin to use all the rest driver as Drivers so I can send Gcodes to them E1, E2?
Using E1 as X2 and E2 as Y2
Thank for any help

Shamoon78, 2020-09-03:
you  want use other drivers as extra axises? or as dual y, dual x, ....?

Shamoon78, 2020-09-03:
As dual y ,x, z the board will configure the e1 as x2 if you enabled in advh in marlin , and e2 as y2 if enabled and go on, if you enable dual y  without enabling dual x marlin will configure e1 as y2, so whatever you enable in the sequence of x2, y2,z2 , will be configured in the sequence of e1,e2,e3,e4 and go on, this senario if you enable one extruder in config.h but if you enable e0,e1 your drivers sequence will start from e2, e3 , etc...

AnHardt, 2020-09-03:
Marlin can currently not handle more than 4 'independent' axes (XYZE) at the same time. So only one E at a time. Only if you have a mixing-nozzle more steppers are in use, but then they have a fixed relation to each other (, the mix). It's mostly the planner what can't handle more. As long the AVRs have to be supported there is no 'convenient' way to do more than 4 axes. We often see attempts to change that, but up to now all of them failed.

bosd, 2020-11-22:
I'm looking for a similar solution, to drive 11 axis independently for CNC purposes.
Sadly, this is not yet possible in Marlin. But there is some interesting development over here:
https://github.com/MarlinFirmware/Marlin/pull/19112

DerAndere1, 2021-08-23:
@bosd, the cited PR was merged into MarlinFirmware/Marlin so Marlin now supports 6 axes plus extruder. If you want even more axes, you are welcome to test https://github.com/DerAndere1/Marlin/tree/9axis_pull and rebase it onto current https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x