bigtreetech/BIGTREETECH-GTR-V1.0 #32: Bootloader of the GTR

P-C-R, 2020-08-24:
Please Provide the Bootloader of the GTR.
Mine got damaged!

Best

GadgetAngel, 2020-09-21:
> Please Provide the Bootloader of the GTR.
> Mine got damaged!
> 
> Best

The GTR V1.0 is based on the SKR PRO board. 

Both the GTR V1.0 board and the SKR PRO V1.1/V1.2 board are STM32F407 processors.  

Look here: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/115 and also look here: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/72#issuecomment-560128508 for fw_stlink.zip.  

Hopes this helps.

P-C-R, 2020-09-21:
> Look here: bigtreetech/BIGTREETECH-SKR-PRO-V1.1#115
> also look here: bigtreetech/BIGTREETECH-SKR-PRO-V1.1#72 (comment)

Nope.they are different!

GadgetAngel, 2020-09-21:
> […](#)

Thanks for letting me know.

GadgetAngel, 2021-01-24:
@P-C-R Here is the bootloader for the GTR V1.0 board.  I use a ST-LINK to download it from my own GTR board:
https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders/GTR%20V1.0

The manual located on the same web site will explain how you can flash the bootloader to the GTR V1.0 board.