bigtreetech/BIGTREETECH-GTR-V1.0 #34: What is the default hardware SPI bus for GTR V1.0 board>

GadgetAngel, 2020-09-21:
I have a GTR V1.0 board with BIQU TMC5160 stepper motor drivers. My printer is Creality Ender-3 Pro.  The motherboard is the GTR V1.0 with TMC5160 on X, Y, Z and E0, I have a V6 hot end and want to attach a PT100 temperature sensor using the Adafruit MAX31865 amplifier board. All other equipment is what came with the Creality Ender-3 Pro. I want to use the Adafruit MAX31865 board and use hardware SPI Protocol in Marlin to communicate between the MAX31865 board and the GTR V1.0 board. I will not be using the K-Type Thermocouple port.

I downloaded the latest bugfix version of Marlin and ONLY made the following changes from the default Marlin for "Creality\Ender-3 Pro" printer (i.e., I copied the default configuration from Creality\Ender-3 Pro\ folder to the Marlin folder). 
Here are the changes I made:

**Changes made to configuration.h**:

```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 3
#define MOTHERBOARD BOARD_BTT_GTR_V1_0
#define TEMP_SENSOR_0 -5
#define HEATER_0_MAXTEMP 400
#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC5160
#define E0_DRIVER_TYPE TMC5160
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
//#define CR10_STOCKDISPLAY

```
**Changes made to configuration_adv.h**:
```
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10
#define X_CURRENT       760
#define X_RSENSE          0.075
#define Y_CURRENT       760
#define Y_RSENSE          0.075
#define Z_CURRENT       760
#define Z_RSENSE          0.075
#define E0_CURRENT      900
#define E0_RSENSE         0.075
#define TMC_USE_SW_SPI
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
```
**Changes to pins_BTT_GTR_V1_0.h file to get the Adafruit MAX31865 board to talk to Marlin via hardware SPI protocol**:
```
#define TEMP_0_PIN                          PF7
/*----------commented out
// SPI for Max6675 or Max31855 Thermocouple
// Uses a separate SPI bus
// If you have a two-way thermocouple, you can customize two THERMO_CSx_PIN pins (x:1~2)

#define THERMO_SCK_PIN                      PI1   // SCK
#define THERMO_DO_PIN                       PI2   // MISO
#define THERMO_CS1_PIN                      PH9   // CS1
#define THERMO_CS2_PIN                      PH2   // CS2

#define MAX6675_SS_PIN            THERMO_CS1_PIN
#define MAX6675_SS2_PIN           THERMO_CS2_PIN
#define MAX6675_SCK_PIN           THERMO_SCK_PIN
#define MAX6675_DO_PIN             THERMO_DO_PIN
*///---------commented out
// define MAX31865 to use hardware SPI
#ifndef MAX31865_CS_PIN
  #define MAX31865_CS_PIN TEMP_0_PIN
  #define MAX6675_SS_PIN MAX31865_CS_PIN   
#endif 
```
**Changes to platformio.ini file:** so that only Adafruit MAX31865 library version 1.1.0 is used:
```
default_envs = BIGTREE_GTR_V1_0
MAX6675_IS_MAX31865     = Adafruit MAX31865 library@1.1.0
```
#### --------------------------Configuration changes ends here---------------------------------------
It all compiles correctly with no warnings or errors.:

[image]

Default hardware SPI bus in variant.h is SPI2 or SPI bus number 2

```
  #define PB12  8 //1:SPI2_NSS / OTG_HS_ID
  #define PB13  9 //1:SPI2_SCK  2:OTG_HS_VBUS
  #define PB14  10 //1:SPI2_MISO / TIM12_CH1 / OTG_HS_DM
  #define PB15  11 //SPI2_MOSI / TIM12_CH2 / OTG_HS_DP
```
```
#define PIN_SPI_MOSI         PB15
#define PIN_SPI_MISO         PB14
#define PIN_SPI_SCK          PB13
#define PIN_SPI_SS           PB12

```

BUT in pins_BTT_GTR_V1_0.h the variant.h SPI pins get overridden and get set to SPI1 or SPI bus number 1.  This is due to the following settings in pins_BTT_GTR_V1_0.h:
```
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION ONBOARD
#endif 
//
// By default the LCD SD (SPI2) is enabled
// Onboard SD is on a completely separate SPI bus, and requires
// overriding pins to access.
//
#if SD_CONNECTION_IS(LCD)
  #define SD_DETECT_PIN             PB10
  #define SDSS                      PB12
#elif SD_CONNECTION_IS(ONBOARD)
  // Instruct the STM32 HAL to override the default SPI pins from the variant.h file
  #define CUSTOM_SPI_PINS
  #define SDSS                      PA4
  #define SS_PIN                    SDSS
  #define SCK_PIN                   PA5
  #define MISO_PIN                  PA6
  #define MOSI_PIN                  PA7
  #define SD_DETECT_PIN             PC4
#elif SD_CONNECTION_IS(CUSTOM_CABLE)
  #define "CUSTOM_CABLE is not a supported SDCARD_CONNECTION for this board"
#endif
```
In variant.h file the PA4, PA5, PA6 and PA7 point to SPI1 or SPI Bus number 1:
```
  #define PA4   39 //NOT FT 1:SPI1_NSS / SPI3_NSS / USART2_CK  2:ADC12_IN4 / DAC_OUT1
  #define PA5   40 //NOT FT 1:SPI1_SCK  2:ADC12_IN5 / DAC_OUT2
  #define PA6   41 //1:SPI1_MISO / TIM13_CH1 / TIM3_CH1  2:ADC12_IN6
  #define PA7   42 //1:SPI1_MOSI / TIM14_CH1 / TIM3_CH2  2:ADC12_IN7
```
I want to wire in a Adafruit MAX31865 board and use hardware SPI not software SPI to communicate with the Adafruit MAX31865 board (Just because I want to do it this way). 

I understand that the Adafruit MAX31865 library does not allow the user to select the SPI bus so it will default to the hardware SPI bus. There is only one default hardware SPI bus for the board.

My question is (with the above set up in firmware) which SPI bus (bus number 1 or bus number 2) is the default hardware SPI bus?

Does the **SDCARD_CONNECTION** to **ONBOARD** changes to SCK_PIN, MISO_PIN an MOSI_PIN affect Marlin globally?

The Variant.h file usually indicates the default hardware SPI bus and on this board it should be SPI 2 but then the pins_BTT_GTR_V1_0.h changes those SPI pins to SPI 1.  **So does it follow that the default SPI hardware bus is now SPI bus number 1?**  

I need to know which bus (bus 1 or bus 2) so I know where to splice into the SPI lines.  If the default hardware SPI bus is bus number 2, I would place a third connector on the flat ribbon cable of EXP2.  If the default hardware SPI bus number is bus number 1 then I will use the BTT TF Cloud V1.0 device and tap into the ESP-12S wireless chip's SPI lines to access the ONBOARD SD card's SPI lines.  So again I ask which bus is the default hardware SPI bus for this situation (bus number 1 or bus number 2)?

GadgetAngel, 2020-11-12:
I answered my own question, see the following link: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/179#issue-741317998

closed 2020-11-12 by GadgetAngel

phongshader, 2020-12-03:
@GadgetAngel can you tell me what your jumper settings are for your TMC5160 drivers?

GadgetAngel, 2020-12-23:
@phongshader you set them for SPI mode.  Look at this manual it might help. https://github.com/GadgetAngel/GTR-V1.0-Stepper-Driver-Jumper-Configuration-Manual/blob/master/CURRENT-Manual/GTR%20V1.0%20Stepper%20Driver%20Jumper%20Configuration%20Version%204.5_reducedFileSize.pdf