bigtreetech/BIGTREETECH-GTR-V1.0 #37: Klipper and gtr

shawnp30, 2020-10-18:
Has anyone successfully compiled gtr with klipper. I get lost after trying to find mcu. I have followed links to use gnd and tx and rx on the li and gtr.  I have disabled serial and usb as klipper and btt suggested. Still cannot find mcu. My pi and board seem to not be communicating what can I do to get this working.

BronsonMMiller, 2021-04-22:
Wow what a wonderful and vibrant community GTR has here. Oct 2020 and not one answer to such a simple question? I should just return this funkkin' board to amazon while I still can...

dzhaparoff, 2021-05-04:
BTT do not care about post sale support... And GTR is not so common used board, community is very small.

shawnp30, 2021-05-04:
My gtr works just fine. It was an error on my part. Gtr is one of the best boards for buck to use. The fact I can run all my motors for corexy triple z in one board with room for an expansion board works perfectly.

closed 2021-05-04 by shawnp30