bigtreetech/BIGTREETECH-GTR-V1.0 #38: jitter on termistor readings on M5 board

Charly333, 2020-11-06:
Got GTR and M5 working together.

GTR has
3x 5160 for XYZ (SPI)
3x 2208 for E123 (UART)

M5 has
3x 2208 for E456 (SPI)
2x 5160 for X2/Z2 (UART)

Termistors on GTR work fine. 
Termistors readings on M5 all have a regulary curve of about 6 degree up and down.
As soon I power up Motors (i.e. Motor drivers, for this happens  with and without motors attached.) this curve becomes totally jitter with uneven peaks and jumps in temperature readings from about 15 degree.
As soon I power down Motor 4 on M5 board with Marlin M18 command - jitter is off again, but altering curve is back again.

Both the altering base curve, as well as the jitter during motor on are much to wide as it could be used for heater controlling in 3d printing.

Swapped motor drivers and termistors between GTR and M5 - all are working fine on GTR - all show fluctuating curves connected to M5.

So how to bring that jitter on M5 thermal readings down? Or at least - what pin to use on GTR to have a termistor for E4 attached to this board while repining in marlin.

Any Idea?

Charly333, 2020-11-08:
Did some additional testing - even if no termistor is attached to PA3 / PF9 / PF10 / PF7 / PF5 - we do have that same jittering behavior on those ports. 

Did us a shielded cable - shield to ground on board side. Little bit dampening but not enough to make it work,

Charly333, 2020-12-13:
OK, found some explanations on termistor errors on other BTT Boards - the problem, the vdda filter is connected to 3.3V different from the thermistor referenced to 5V.

https://github.com/KevinOConnor/klipper/issues/2194#issuecomment-622667850

Seems to be the same on M5, for even they have drawn as much onto 3.3 volt as possible. That differs from former BTT SKR boards.

DigDagGoo, 2021-02-25:
@Charly333

i have exactly the same unsatisfactory situation. The sensors on the GTR work perfectly, the sensors on the M5 board do not.

But I can't confirm that the VCC is different. In both cases this is taken from 3.3V (in the diagram VCC_3.3V), also at pin2 on the 50-pin connector. There is only L1 in between to VREF on the ARM. All other consumers then go through the R61 (0 Ohm). I only see a small difference with GND. The GTR board uses GNDA0 and the M5 board uses GND - in between there is also a zero ohm resistance (R93).

But with all of this information, I still have significant differences. Temperature jumps of the GTR board +/- 0.2 ° C on the M5 board +/- 1.5 ° C ....

I also recreated the resistor divider on a breadboard (without the protection diodes but with the capacitors) that didn't help either. I used GNDA0 and VCC_3.3V. That didn't get any better either.

Did you find a solution to solve the problem?

If my "google-English" is not understandable, please ask.