bigtreetech/BIGTREETECH-GTR-V1.0 #40: [GUIDE] to using PT100 or PT1000 or Thermocouples with SKR PRO V1.x boards or GTR V1.0 boards

GadgetAngel, 2020-11-12:
# GUIDE:

1. You CAN NOT mix PT100/PT1000 and thermocouples:
 
If you have only one hotend then you can have the following:
- Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0
- PT100 (-5) on E0
- PT1000 (-5) on E0
- Thermocouple (-3) on E0 

If you have two hotends then you can have the following:
- Thermistor (that is an ADC Temp Sensors [uses a pure analog I/O pin]) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
        Examples of this are:
               - PT100 (uses table 20 or 21) on E0 and Thermocouple (-4 - use pure analog pin) on E1
               - Thermocouple (-4 [pure analog I/O pin]) on E0 and PT100 (uses table 20 or 21) on E1
              - PT1000 (uses table 1047) on E0 and PT100 (uses table 20 or 21) on E1
              - too many to enumerate (etc.)
- PT100 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT1000 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure  analog I/O pin]) on E1
- Thermocouple (-3) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT100 (-5) on E0 and PT100 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT1000 (-5) on E0 and PT1000 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 IS ALLOWED ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- Thermocouple (-3) on E0 and Thermocouple (-3) on E1===>>> can use either Software SPI or Hardware SPI.  They both work for two MAX31855 boards and two thermocouples

If you have two hot-ends then you **CAN NOT have the following**:
- PT100(-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- PT1000 (-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT100 (-5) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT1000 (-5) on E1 IS NOT ALLOWED
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT100 (-5) on E1 IS NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT1000 (-5) on E1 is NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and Thermocouple (-3) on E1 is NOT ALLOWED ***

*** If you only have ONE PT100 (or PT1000 or Thermocouple) on your printer and you have two hotends then in Marlin you
must use TEMP_SENSOR_0 as the PT100 (or PT1000 or Thermocouple) even if you have the PT100 (or PT1000 or Thermocouple) hooked up to the second extruder.  Just override your PIN definitions to have TEMP_0_PIN point to where the PT100 (or PT1000 or Thermocouple) is located. 

---

2. Marlin still uses the old naming scheme of the 20x4 display `REPRAP_DISCOUNT_SMART_CONTROLLER` in the pins-file for the `software SPI pins names`:
The EXP1 and EXP2 ports are used for display screens.  EXP1 contains a HARDWARE SPI bus while EXP2 contains a Software SPI bus.  The EXP1 hardware SPI bus signal names are obvious in the pins-file but the EXP1 software SPI bus signal names are **NOT obvious**. Here is the EXP1 software SPI bus signal names used in pins-file with their corresponding SPI software functions:
```
LCD_PINS_D4 is software SPI signal SCK for EXP1
LCD_PINS_ENABLE is the data line or software SPI signal MOSI for EXP1
LCD_PINS_RS is Register select (command/data) or software SPI signal SS for EXP1
```

---

3. Translation table to be used (or Thermistor table) depends on the power you supply to the amplifier board and the ADC reference voltage of the MCU:

- 1. Use table 20: When the power supply for the PT100/PT1000 Amplifier board is equal to the ADC reference voltage for the MCU.
--  For SKR PRO V1.1/V1.2, SKR V1.3/V1.4/V1.4 Turbo, GTR V1.0, M5 V1.0 all use 3.3 VDC as the ADC reference voltage for the MCU.  Therefore use table 20 when the PT100/PT1000 Amplifier board is powered by 3.3VDC
- 2. Use table 21: When the power supply for the PT100/PT1000 Amplifier board is different to the ADC reference voltage for the MCU. 
-- The E3D PT100 Amplifier board wants 5V as its power supply. But the GTR V1.0 (like the SKR PRO) uses 3.3V as the ADC reference voltage for the MCU.  If you use 5VDC power to the PT100 amplifier board, the amplifier can output a maximum signal with 5VDC.  To protect the GTR V1.0  board (or the SKR PRO board), use a 3.6V Zener Diode (reverse biased hookup) between the output signal of the amplifier board and ground (to protect the MCU from getting fried from a 5V input).

[image]

On the GTR V1.0 board, here is a list of PINS that you will need to prevent negative current injection. We prevent negative current from flowing into the MCU by using a Schottky diode between the output signal of the amplifier board and ground.  I do this as preventive measure. It protects the MCU from getting fried.

[image]
[image]

The below wiring diagram for PT100 using Analog ADC input using 5V DC for the PT100 amplifier board but the MCU board (GTR V1.0) uses 3.3 VDC ADC reference voltage, therefore the Thermistor table to use for this is Table 21:

[image]

The below wiring diagram for PT100 using Analog ADC input using 3.3 VDC for the PT100 amplifier board but the MCU board (GTR V1.0 board) uses 3.3 VDC ADC reference voltage, therefore the Thermistor table to use for this is Table 20 (see the process data sheet and you will find that PF10 need protection against negative current injection,https://www.st.com/resource/en/datasheet/stm32f407ig.pdf#page=114):

[... 497 lines cut ...]
- If you have a **MCU like the GTR V1.0 board**, which already has a MAX31855 built into the board (I will call this ONBOARD_MAX31855), then to use the **GTR V1.0 board with an external MAX31855** you MUST use the Clock Line (SCK) and Data Output Line (DO)  of the ONBOARD_MAX31855 PINs defined in **pins_BTT_GTR_V1_0.h** file. This would mean you have to tie the second (external) MAX31855 amplifier board to the M5 connector of the GTR V1.0 board so that you gain access to the DO and SCK lines of the ONBOARD_MAX31855 chip.

For example, here is how you would **setup using two thermocouples with the GTR V1.0 board**:
```
#define TEMP_0_PIN   PH9    //CS for ONBOARD MAX31855 chip, See NOTE below
#define TEMP_1_PIN   PH13  //CS for external MAX31855 amplifier board
#ifndef MAX6675_SS_PIN
     #define MAX6675_SS_PIN            TEMP_0_PIN     //for first Thermocouple
     #define MAX6675_SS2_PIN           TEMP_1_PIN    //for second Thermocouple 
     #define MAX6675_SCK_PIN           PI1
     #define MAX6675_DO_PIN            PI2
#endif
```
**In configuration.h**:
`	set TEMP_SENSOR_0 to -3` and `set TEMP_SENSOR_1 to -3`

**NOTE: the first thermocouple is wired to the ONBOARD K_Type Thermocouple (KTEM) connector.  That KTEM connector by default uses PH9 as its Chip select pin.  The first thermocouple is not shown in the wiring diagram.**

Here is a wire diagram for this situation:

[image]

---

END OF SUMMARY

truglodite, 2021-06-17:
Note that #6 is incorrect as Marlin actually does support 3 wire Rtd. I have a working example of a 3 wire pt100 with the max31865 and Marlin 2.0.8.1, on ramps 1.4 with mega2560.

Otoh, with a Gtr 1.0 I am having trouble getting the max31865 to work at all using info in this guide. I have been testing with a bare Gtr board (no 12V power or max wired up), and the terminal freezes requiring reset shortly after the UBL messages appear after boot. This happens with 2.0.8.x. I submitted and issue for 2.0.9 because that won’t even compile when -5 is set as a sensor.

truglodite, 2021-06-21:
So I did manage to get this all working with my GTR. Looks like something happened that broke it all after Marlin b2bd547a275109b01e826b351f1e6066b2e23ebb. I tried several others and found that 2072 works, but gives lots of uneeded errors in serial. 208 and up do not seem to work at all (always max temp errors at boot up). Otherwise, this guide more or less helped. My config changes were a little simpler. TEMP_SENSOR_0 does have to be declared (I also used PI9 for cs there).

I used hardware SPI with onboard SDcard. So I just declaring the max6755 cs pin, and used a full size card adapter that has dupont spaced pins for easy soldering. I did have issues initially using the MAX sensor with sdcard. I found out it was just the sdcard at fault. The 128MB micro sd that came with my GTR will cause a reset every time it is accessed. I replaced with a spare 64gb samsung high reliability micro sd card (formatted just 4gb fat), and it all works great now. Not sure what was up with the sdcard, but found another issue report on marlin that mentioned a bad sd card adapter was at play. In my case it was actually the sdcard itself. So if anyone has issues getting this to work, make sure to pull out the sd card and adapter for testing.