bigtreetech/BIGTREETECH-GTR-V1.0 #41: BLTouch configuration

Decalis1979, 2020-11-22:
I am unable to configure my BLtouch on the GTR. It works by advancing and retracting the pin. How much will the reading of the pin go forwards and backwards several times until the error appears. The same is true when searching for Home in Z. Please does anyone have a solution?
Thanks

Lifeforc, 2020-12-04:
Change pin in pin file to PH11  go to configuration. H file search for probe pin change that PIN to PH11

dixi83, 2020-12-22:
search for `Z_MIN_PROBE_PIN` set it to `PH11`

bholota, 2021-02-13:
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN PH11
#define BLTOUCH
```
works for me

Decalis1979, 2021-06-21:
Please am trying to update the firmware of my TFT-70. But I'm not getting
it.
It's giving an error when compiling:
Can you help me please

[image: image.png]

Red error:    collect2.exe: error: ld returned 1 exit status***
[.pio\build\BIGTREE_TFT70_V3_0\BIGTREE_TFT70_V3.0.27.x] Error 1

Thanks

closed 2021-07-06 by Decalis1979

Lifeforc, 2021-07-06:
You don't need to compile copy proper folders from firmware download to sd
and put in tft and turn on

Decalis1979, 2021-07-06:
Hi thanks for the reply, on 02/07/21 I managed to solve this problem. But
now I'm having two issues with Marlin 2.0.9.1:

1st Problem:

 TFT does not open Marlin's screen (I configured as below)

#####################################################################
// @section machine

/**
 * Select the serial port on the board to use for communication with the
host.
 * This allows the connection of wireless adapters (for instance) to
non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino
bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Serial Port Baud Rate
 * This is the default communication speed for all serial ports.
 * Set the baud rate defaults for additional serial ports below.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 *:[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
//#define BAUD_RATE_GCODE // Enable G-code M575 to set the baud rate

/**
 * Select a secondary serial port on the board to use for communication
with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3
#define BAUDRATE_2 115200 // Enable to override BAUDRATE

/**
 * Select a third serial port on the board to use for communication with
the host.
 * Currently only supported for AVR, DUE, LPC1768/9 and STM32/STM32F1
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_3 1
//#define BAUDRATE_3 250000 // Enable to override BAUDRATE

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH
############################################################################

2nd Problem:

I select on the display the extruders E0, E1 and E2 but only the E0 moves,
the E1 and E2 do not move.

Can you help me please?

Thanks

Lifeforc, 2021-07-06:
What board or machine are you using

Lifeforc, 2021-07-06:
If it's gtr ok.  Did you set to the hotends in marlin