bigtreetech/BIGTREETECH-GTR-V1.0 #45: Upload board firmware without removing onboard SD

dixi83, 2020-12-22:
Hi there,

I am upgrading from a SKR 1.3 to the GTR 1.0. One of the first things I noticed is that the only way to load the firmware to the board is to remove the SD card put it in a cardreader and copy the 'firmware.bin' to the root. I am experimenting now with the features I have in mind for the other upgrades. Wich takes a lot of SD In and Out... Because it is already build in (with minimum space of course ;) ) I am afraid I'll break the card or the cardholder...

With the SKR 1.3 when I connect the USB to the mainboard, the SD card gets mounted in as an external drive. PlatformIO automatically puts it on the mounted drive, reset the board and you're ready to test the changes. 

Is the something like this possible with the with the GTR? Or maybe over a serial connection?

Thanks, Martijn

dixi83, 2020-12-24:
OK, I did some research and my conclusion is that it will be possible by adjusting the the marlin firmware. You have to adjust the boot loader... since the board has the same CPU as the BTT SKR Pro. I'll give it a try to flash the modified bootloader by https://github.com/Arksine/STM32_HID_Bootloader/releases/tag/v0.7
What I see now i have a few challenges:
- Make a backup of the current bootloader.
- Wire out the `BOOT1` pin

The `BOOT1` has jumpers on the SKR Pro, but on our beloved GTR it has only a resistor (R9) to ground:
[image]
which is located over here:
[image]

I ordered a ST-Link om amazon when it arrives I'll give it a try and let you know the results.

Mikusluz, 2020-12-29:
That would be excellent if it works out!

DrumClock, 2021-01-12:
Hi @dixi83 
What does it look like?  It works ?

GadgetAngel, 2021-01-14:
I also am curious @dixi83 does your idea work?

dixi83, 2021-01-15:
Hi there I tried flashing the firmware, I managed to get stuf working but you'll have to make a 3.3v connection to the boot1 pin and reset the board every time you'll have to upload new firmware... not as efficient as I hoped... so I dug a little deeper in uploading firmware to ST32 chips. And I found that the by far easiest way is to just use the STlink to this port: 
[image]
I made an easy pluggable cable for it:
[image]

and adjusted my platformio.ini file like this:
```[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_GTR_V1_0_stlink
include_dir  = Marlin
```

and added this below the `[env:BIGTREE_GTR_V1_0]` section
```[env:BIGTREE_GTR_V1_0_stlink]
platform          = ${common_stm32.platform}
extends           = common_stm32
board             = BigTree_GTR_v1
upload_protocol   = stlink
extra_scripts     = ${common.extra_scripts}
  pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags       = ${common_stm32.build_flags}
  -DSTM32F407IX -DVECT_TAB_OFFSET=0x8000```

hope this hels you guys

closed 2021-01-15 by dixi83

truglodite, 2021-06-17:
@dixi83 have you tried this with marlin 2.0.8 or later? 

I just got my stlink to work on my GTR today, but the platformio.ini file looks like it has changed since you wrote this, and I can't seem to get your code changes to work. The board env: sections have all dissapeared. Copy/pasting your changes to the default platforio.ini now gives an error:

`Error: Unknown board ID 'BigTree_GTR_v1'`

Has anyone got stlink working for flashing a GTR with Marlin 2.0.8+?

Any help is much appreciated... my GTR is giving me some stink, and I doubt the onboard microsd slot will survive much more debugging.