bigtreetech/BIGTREETECH-GTR-V1.0 #49: [HELP!!!] BTT GTR v1.0 and Raspberry Pi 3 B+ do not connect

DEAth1981, 2021-01-06:
Good day!

Available:
- BTT GTR v1.0;
- Raspberry Pi 3 B+.

Did:
1. Connected pins (GPIO14, GPIO15, GND) on Raspberry Pi 3 B+ with pins (PA9, PA10, GND) on BTT GTR v1.0.
2. Installed OctoPrint and Klipper on Raspberry Pi 3 B+.
3. Compiled the firmware for BTT GTR v1.0 according to the screenshot:
[image]
4. Copied "klipper.bin" to MicroSD BTT GTR v1.0, before renaming the file to "FIRMWARE.CUR".
5. Created "printer.cfg" using "generic-bigtreetech-gtr.cfg" as an example
6. In Raspberry Pi 3 B+ disabled Linux serial console and got free "/dev/ttyS0".
7. In the connection settings OctoPrint set the serial port "/tmp/printer" and the baud rate "250000".
8. Rebooted BTT GTR v1.0 and Raspberry Pi 3 B+.

What's happening:
1. In OctoPrint, I click the "Connect" button and get the "Klipper: Connected to host" status.
2. I am trying to do something on the printer, for example, shuffle along the X axis.
3. I get errors with different frequency:
- "Can not update MCU 'mcu' config as it is shutdown";
- "Must home axis first: 10.000 0.000 0.000 [0.000]";
- "Shutdown due to M112 command".
4. After any error appears, I get the status "Klipper: Disconnected from host".

What am I doing wrong? How do you achieve stable operation of this bundle on Klipper?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5772377/klippy.log)

DEAth1981, 2021-01-06:
I was already so worried that I decided to disconnect communication via GPIO and connected via USB, i.e. Pi -> USB cable -> GTR.
Recompiled firmware with USB enabled.
I fixed the configuration to use the new port, but nothing changed - ERRORS REMAINED!
I'm starting to hate this GTR, and at the same time the manufacturer BTT for the lack of any support !!!

DEAth1981, 2021-01-07:
So I could not get this "set" to connect via UART. I ended up connecting the board and Pi via a USB host on the board. Everything works now.
I think that the problem was in the Raspberry Pi 3 B+, or rather in the peculiarities of its two UART ports... it was necessary to buy a Raspberry Pi 4 B.

closed 2021-01-07 by DEAth1981

Protean-Man, 2021-03-18:
Only connect the GTR and Pi using RX/TX/GND to GPIO pins.
You do not have to change the GPIO settings in the Pi. Just flash Octoprint and go.
I am also struggling with the GTR and Pi with some specific communication issues (Pid Autotune). 

In Marlin set these two as below and then flash the board. This should fix your communication issue. You can try other ports and let me know what works if for some reason this does not. 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

Also, I think many people are having issues with the GTR and Klipper. There may be a communication block in the design.

DEAth1981, 2021-03-18:
> Only connect the GTR and Pi using RX/TX/GND to GPIO pins.
> You do not have to change the GPIO settings in the Pi. Just flash Octoprint and go.
> I am also struggling with the GTR and Pi with some [...]

What does your post have to do with my problem? Or is it just agitation to use Marlin and not Klipper?
And to mine from my first post it is more than clear that your advice to use GPIO for communication does not work for me.

P.S. Sometimes it is better to chew than to speak (in this case, write... the Russians will understand). :)