bigtreetech/BIGTREETECH-GTR-V1.0 #51: Upload board firmware without removing onboard SD

GadgetAngel, 2021-01-15:
This is a continuation of #45, I got to believe that someone with knowledge of how boot loaders work can figure out how to add the USB connection for the SKR PRO V1.x board or the GTR V1.0 board.

If someone can add USB Storage, Serial, DFU and WebUSB interfaces to a BLUE PILL device then why can it not be added for the STM32F407IX processor?

I found this on the web: https://medium.com/coinmonks/stm32-blue-pill-usb-bootloader-how-i-fixed-the-usb-storage-serial-dfu-and-webusb-interfaces-36d7fe245b5c

Along with https://github.com/Arksine/STM32_HID_Bootloader

Can someone figure out how to combine the USB Storage, Serial, DFU and WebUSB to github.com/Arksine/STM32_HID_Bootloader project?

Well I am going to try, I will post what my findings are here.  I also found this resource: https://github.com/KevinOConnor/klipper/blob/master/docs/Bootloaders.md

I am not an embedded software engineer so all this is new to me.  Is there someone out there that is more experienced in this area that can answer my question, why can we not add WebUSB to the GTR V1.0 bootloader or to the SKR PRO V1.x bootloader?  Both boards have plenty of flash space and I would not mind giving up flash space for a user bootloader that is developed that would allow me to update my firmware by using the Chrome browser

DrumClock, 2021-01-15:
Hi, I think if they wanted to in BTT, it wouldn't be a problem for them.
Unfortunately they don't want to ..... they won't even provide the original bootloader ( if you delete it :) )

GadgetAngel, 2021-01-17:
@DrumClock from what I am learning at least on the SKR V1.x board the boot loader resides from 0x08000000 - 0x08007FFF so your compiled firmware starts at address 0x08008000.

From looking at the platformio.ini file it looks like the GTR V1.0 board board also has a bootloader from 0x08000000 - 0x08007FFF so your compiled firmware start at address 0x08008000.

I have both boards so I can backup the boot loader with firmware and save it on my boot loader repository just in case I over write it.  I will save the boot loader and the firmware+bootloader file on my repository located here: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader

Then I will attempt to try and figure out a way to make all this work but it will take some time.

GadgetAngel, 2021-01-22:
I already own or have purchased all the BIGTEETECH boards **except for ONE**: **SKR min E3 V1.0, does anyone know where I can get my hands on one?**

This repository will have all the bootloaders for all BIGTEETECH boards back up in this directory: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders

I have written manuals on how others can do this for their boards, they are written in .pdf format.

I need to zip all the binary and hex files up, but I will do that later.

I need to write a manual on how to rip the bootloader and bootloader+firmware on a Linux machine. I am presently working on that manual.

I am waiting on the following boards to arrive: SKR mini V1.1 and SKR CR6.  I can not get hold of the SKR mini E3 V1.0 board

I will finish the manual about the how to backup and reload the bootloader on Linux using (st-flash, st-info tools and openocd, and telnet tools)  this week.

So if anyone knows how I can get my hands on an SKR mini E3 V1.0 board, I wish they would tell me.

LightTribe-art, 2021-01-23:
I'm configuring a GTR for a custom printer and having to remove the sd card is a nightmare. Thanks for working on this.

GadgetAngel, 2021-01-23:
@LightTribe-art I am still learning about all this so it will not take some time.  I agree with you tho, having to update via SD card is not really necessary. If they have found a way to upload the app file via chrome for the blue pill, I believe the software to do it for the printer boards is there I just need to pull it together and make it work.

davhensl, 2021-01-31:
@GadgetAngel : Thank you for all your work; very tedious (among other adjectives) what you've done, but you've probably rescued 1000's of users of BTT boards! Well 100's in any case. BTT should sponsor you, or at least provide you with some equipment, as you are providing them with a very useful service, as is all the Marlin team of course.

I recently had the dying sdreader issue on a brand new SKR 1.4 Turbo. I used the BTT-Writer + Flashmagic and successfully flashed some of the firmwares from both you and others, however NONE of them would allow me to boot into a workable state as they all had some config parameter with unintended consequences which caused the board to HALT (mintemp E1 error, etc) when booted with nothing plugged in other than the TFT/LCD. One does this to isolate back voltage/back power/shorts for example when debugging etc.

After a few weeks of email, BigTreeTech replaced my board (arrived Sat 1/30/2021) and I backed up the firmware, bootloader, and eeprom memory contents, as the board booted fine with no peripherals attached.  These, when flashed to my previous broken board, allowed it to boot with no errors or halts, allowing me to connect with Pronterface/Printrun and continue working with it. Would you be interested in having copies of these backup/dumps, and if so, what is the best way to get them to you?
I noticed that your DFU github page has no issues, so I didn't make one there...seems like your project is popular and useful enough that perhaps we should start using the issues section?

Really, thanks again for all your work!

GadgetAngel, 2021-02-01:
> @GadgetAngel : Thank you for all your work; very tedious (among other adjectives) what you've done, but you've probably rescued 1000's of users of BTT boards! Well 100's in any case. BTT should [...]

 I would be interested in getting your dumps from the SKR V1.4 Turbo board. I am only interested in the dumps from the SKR V1.4 Turbo as it arrived (before you made changes to your firmware).  I want the bootloader-and-firmware.bin file and the same in ihex format: bootloader-and-firmware.hex files.  Do you have a copy of the micro SD card contents before you used the micro SD card to update the firmware? If so, I would like a copy of those contents also.

This is how you get it to me.  Binary files and Hex files are best transported in .zip folder. Label the folder SKR-V1.4-Turbo-bootloader-and-firmware-ORIGINAL.zip. The contents of the SD card that came with the board from BIGTREETECH label as Origianl-SD-contents.zip 

Place the bootloader-and-firmware.bin and bootloader-and-firmware.hex file into "SKR-V1.4-Turbo-bootloader-and-firmware-ORIGINAL" folder then use ZIP utility to make the .zip file.

Place the contents of the SD card into the folder "Origianl-SD-contents" then use the ZIP utility to make the .zip file.  

Now use your free google drive, and upload the SKR-V1.4-Turbo-bootloader-and-firmware-ORIGINAL.zip and Origianl-SD-contents.zip to your free google drive.

Turn on the google's internet share link to both uploaded .zip files now on your google drive.  Then copy the shared link to your clipboard and place it into a response to this message.  Copy the shared link to your clipboard for the other .zip file and again place it into the response to this message.  

Now I will be able to download your .zip files from your google drive to my computer and place them up on the web site.

davhensl, 2021-02-01:
@GadgetAngel 
As you have written, so shall it be done! Within some definition of "margin of error."

I have not updated the firmware on this board yet, so the SD card file FIRMWARE.CUR has not been altered. The only operations run on this board have been informative reports (M122, M119, M115, M503, etc) and initial M502, M500, M504, M501, and M503.

:s/Origianl/Original/
[Original-SD-contents.zip](https://drive.google.com/file/d/1vQ91gx7GaR4W0QreoQjElqimVfH2Oozk/view?usp=sharing)

 From BTT TFT70 V3.0 Touchscreen mode "Info" screen, firmware section once loaded:
   System: Marlin 2.0.5 (GitHub)
   Machine: 3D Printer
(LCDSimulator/"Marlin mode" didn't load anything so I can't report what firmware version it sees)

 SD contents as listed in Windows10 explorer:
   FIRMWARE.CUR
   Size:                156 KB (160,320 bytes)
   Size on disk:   158 KB (161,792 bytes)
   Created:          Tuesday, December 15, 2020, 10:34:19 PM
   Modified:        Tuesday, April 21, 2020, 6:57:02 PM

From FlashMagic "View file Information":
  Last modified: 2/1/2021 1:31:03 AM
  File size: 450964 bytes 
  MD5 hash: 856FAE52C919B0A40E1521C7ADB84583

  Memory 0x00000000 : 0x0007FFFF: 30.58% used
   Lowest address used: 0x00000000
   Highest address used: 0x0002723F
   Data range: 160320 bytes (including gaps)
   CRC32 of data: 0x8E4D12BB (polynomial 0x82608EDB)

[SKR-V1.4-Turbo-bootloader-and-firmware-ORIGINAL.zip](https://drive.google.com/file/d/1X10oQhybQyQ2tV8R97O9o1IMMofhx9Xi/view?usp=sharing)

All dumps from Flashmagic ran with the "Verify after Programming" option enabled with no errors reported.
From Flashmagic "ISP...Read Signature":
Signature:	0x26113F37
Bootloader Version:	4.2

From Flashmagic file information:
  Last modified: 1/31/2021 2:25:36 PM
  File size: 1474709 bytes 
  MD5 hash: 20672B05389777065ABA77F3FF1AFE17

  Memory 0x00000000 : 0x0007FFFF: 100.00% used
   Lowest address used: 0x00000000
   Highest address used: 0x0007FFFF
   Data range: 524288 bytes (including gaps)
   CRC32 of data: 0x2EE8CA4B (polynomial 0x82608EDB)

GadgetAngel, 2021-02-01:
> @GadgetAngel
> As you have written, so shall it be done! Within some definition of "margin of error."
>
> I have not updated the firmware on this board yet, so the SD card file FIRMWARE.CUR has not been [...]

Thank you!

DrumClock, 2021-02-13:
@GadgetAngel 
can you please help me to solve this problem with **MKS Rumba32** board?
https://github.com/makerbase-mks/MKS-RUMBA32/issues/12#issuecomment-778581558

truglodite, 2021-06-17:
I apologize for the 2xpost... was not sure with the other issue closed anyone would see it. So I'm referencing here:

https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/issues/45#issuecomment-863606965

I admit, I need some further hand holding to get pio to flash my gtr with marlin 2.0.8 using my stlink. I am guessing you guys are working with bugfix right? Anyone care to share their platformio.ini changes for flashing with stlink?

GadgetAngel, 2021-06-18:
> I apologize for the 2xpost... was not sure with the other issue closed anyone would see it. So I'm referencing here:
> 
> [#45 (comment)](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/issues/45#issuecomment-863606965)
> 
> I admit, I need some further hand holding to get pio to flash my gtr with marlin 2.0.8 using my stlink. I am guessing you guys are working with bugfix right? Anyone care to share their platformio.ini changes for flashing with stlink?

By design the GTR board does NOT use the upload firmware from VScode's PIO command.  The firmware must be rewritten and I have gotten too busy to work on it.  

To upload new firmware for the GTR board one must manually add the new marlin firmware to a micro-SD card and then place it into the GTR board's micro-SD card reader and manually reboot the board (or cycle power on your 3D printer).

zeleps, 2022-02-06:
Hi everyone, just uploaded this: https://github.com/zeleps/BootloaderZ. Requires an STLink device to upload it. Once configured, built and uploaded, you can use a trigger pin (default is LCD button) while resetting to enter the USB HID flash mode. Works fine with PlatformIO's `upload_protocol = hid`. Should work with any STM32F4xx board, but it has been tested only with BTT GTR 1.0 so far.

Since I haven't written any instructions yet, I want to stress out that you should be *absolutely certain* that you have a copy of your original bootloader (or can get it from @GadgetAngel's repo) which you can flash back to your board, before trying this one. It has not been widely tested and it might not work as expected.

zeleps, 2022-02-13:
In case anyone is still interested in this, I added DFU support to my bootloader project.