bigtreetech/BIGTREETECH-GTR-V1.0 #53: Dual Z drives

MartinDavids, 2021-01-31:
Has anyone use E1 as a second Z drive? if so can you help me with what you changed in Marlin to get it to work. 

Thanks in Advance

ediyouri, 2021-02-09:
I've configured my GTR like this and it is working great.

Changes in config.h
```
define X_DRIVER_TYPE  TMC2209#
define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2209
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209
```

Changes in config_adv.h
```
#define NUM_Z_STEPPER_DRIVERS 2

[...]

#define Z_STEPPER_AUTO_ALIGN
```

Make sure to put the jumper back into the Z2 motor pins and use E1 instead.

DocYester, 2021-05-28:
So it should be possible to use this board for a triple-z setup ?!
Just need X,Y,Z1,Z2,Z3,E
first Z goes into Z1 Port , second in E0 , third in E1 and the E-Motor in E2

geozoc, 2021-06-02:
Yes triple-z would be possible if a single head extruder is used otherwise you would need to upgrade the GTR with the extension board M5.

Regarding the setup above: I suggest to also add a defininition for the second Z-endstop or take care which one is used.

Best G

melslenstra, 2021-12-27:
> So it should be possible to use this board for a triple-z setup ?! Just need X,Y,Z1,Z2,Z3,E first Z goes into Z1 Port , second in E0 , third in E1 and the E-Motor in E2

I think you got the order wrong; Marlin first assigns extruders and then additional Z channels to the unused E_n_ channels. So in your case, you'd end up getting:
```
X = X
Y = Y
Z1 = Z
E = E0
Z2 = E1
Z3 = E2
```

DocYester, 2021-12-27:
Order isnt really relevant at all ...
Made it with 4 Z-Axis X , Y , E , Z , Z1 , Z2 , Z3
The "doubled" Port for 2 Z Axis is now my Extruder ... just to have it clean

melslenstra, 2021-12-27:
How did you achieve that? Did you remap the relevant pins in the `pins_XXX.h` file for the board?

DocYester, 2021-12-28:
> How did you achieve that? Did you remap the relevant pins in the `pins_XXX.h` file for the board?

Yes