bigtreetech/BIGTREETECH-GTR-V1.0 #58: GTR V1.0 and BTT TMC2208 Error:ALL LOW

edouglas37, 2021-03-15:
So i have a BTT GTR v1.0 and 6 BTT TMC 2208's V3.0 UART's

Now the problem is . 4 test with no errors 2 test with errors. 

Firmware: Marlin
Version: CONFIGURATION_H_VERSION 020008

17:05:37.978 : Testing X connection... OK
17:05:37.997 : Testing Y connection... Error: All LOW
17:05:38.006 : Testing Z connection... OK
17:05:38.025 : Testing E connection... Error: All LOW
17:05:38.034 : Testing E1 connection... OK
17:05:38.042 : Testing E2 connection... OK

I have tried Swapping the steppers but the X and E0 still have the same error so its not the steppers its either something on the board or something in the firmware.

hiperlouro, 2021-03-18:
I'm having the same issue with my BTT GTR 1.0 and BTT TMC 2208 V3 UART.

But my issue is only on the X axis. 

I've also tried swapping drivers but no success.

Here's my TMC DEBUG Output:

SENDING:M122
axis:pwm_scale/curr_scale|flags|warncount
		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	795	795	795
MAX current	2025	1121	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	0/31	12/31	12/31	12/31
PWM scale
vsense		0=.325	1=.18	1=.18	1=.18
stealthChop	false	true	true	true
msteps		256	16	16	16
interp		false	true	true	true
tstep		0	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	0	17	15	34
pwm scale auto	0	0	0	0
pwm offset auto	0	42	38	86
pwm grad auto	0	14	14	14
off time	0	3	3	3
blank time	16	24	24	24
hysteresis
 -end		-3	-1	-1	-1
 -start		1	1	1	1
Stallguard thrs
uStep count	0	40	40	72
DRVSTATUS	X	Y	Z	E
sg_result
stst		*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

edouglas37, 2021-03-18:
@hiperlouro i changed boards, then swapped the drivers around and got x to read normal. I think it might be a stepper issue ?

BirkBinnard, 2021-03-20:
I've received the ALL LOW error message every time I boot my BTT GTR1.0 board that is running Marlin 2.0.7.2. But so far the board seems to function OK. It has TMC2208 stepper drivers installed. I've done everything I can think of to find or identify the cause of this error message, but have found neither a cause nor a solution. What I have found is that people have been seeing this error for at least a couple of years.

Today I used VSCode to search all of Marlin (and maybe PlatformIO too - I'm not sure how VSCode does it's searches) for the text "ALL LOW" and found nothing. So maybe the error is coming from somewhere else? I'm using a BTT TFT70 screen to communicate with the GTR 1.0, so maybe there is something going on with it - the TFT 70.