bigtreetech/BIGTREETECH-GTR-V1.0 #59: Board stopped connecting to USB/R-Pi

Protean-Man, 2021-03-19:
Anyone else having issues with serial port giving out on this board? 
I have been running this thing for a few months, testing various components. Run the GTR with a Raspberry Pi 4 and through USB to Pronterface - Both working fine (except PID Autotune is a disaster and fails out of the box).

Today I turned on the printer and the GTR just stopped working. Nothing changed.
Tried disconnecting the Raspberry Pi and running just USB and still nothing. 
Recompiled firmware to the SD, and this won't even load in. The flashing green light when loading new firmware doesn't run. 

Anyone else having this issue? I honestly have no idea where to look or what to fix. Scanned the board and there aren't any burnt components either, with no smell. 

Thanks!

ZGMFX20AR, 2021-03-20:
I've got the same issue, this happen after I made some changes on the thermistor section on marlin and recompiled it. Even though the compiling process was success, the terminal shows some warning regarding the LiquidCrystal.

`In file included from .pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal\FastIO.cpp:43:
.pio\libdeps\BIGTREE_GTR_V1_0\LiquidCrystal\FastIO.h:217:106: warning: 'boolean' is deprecated [-Wdeprecated-declarations]`

When I flash the firmware,  the board was not detected and unable to establish connection with Pronterface as well as Octoprint. 

Have you got any backup to your earlier firmware? as currently I'm using the older version of the firmware binary file that I've saved earlier and the board working again. 

But its frustrating me enough as I wont be able to made changing on the firmware for the time being.

Protean-Man, 2021-03-20:
Ok you've helped me find this:
```
DEPRECATED! A legacy library storage `C:\Users\----\OneDrive\Documents\GitHub\Marlin\.piolibdeps` has been found in a project. 
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `C:\Users\----\OneDrive\Documents\GitHub\Marlin\.piolibdeps` folder.
More details -> https://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing BIGTREE_GTR_V1_0 (platform: ststm32@~10.0; board: BigTree_GTR_v1; framework: arduino)
```
I'll try to figure this out and post what I find.

How were you able to upload the previous version of your firmware if the GTR doesn't even read a firmware change on the SD Card? At least that is what I think it is doing. When I change the firmware and restart the GTR the firmware LED doesn't show any change (blinking green LED).

Protean-Man, 2021-03-20:
I think I found out why. 
The 3.3V on the board stopped for some reason. Some kind of short circuit. I'm going to have to find the component that broke and see if I can replace it. Shouldn't be hard - just a SMD resistor or something.

Shamoon78, 2021-04-03:
The chip colored in yellow is responsible for requlating 3.3 volt , you can find it easly in adrino old board.
(https://user-images.githubusercontent.com/60273204/113486098-5f7eab00-94f4-11eb-8f1b-047ab986a6fa.JPG)

DEAth1981, 2021-05-31:
> Anyone else having issues with serial port giving out on this board?
> I have been running this thing for a few months, testing various components. Run the GTR with a Raspberry Pi 4 and through USB to [...]

Likewise... I turned on the printer, there is no connection between RPi and the MCU, the firmware is not installed... the multimeter on the contacts of the connectors of the limit switches shows about 380mV instead of 3.3V.

P.S. The board is new, it has been in operation for about 3 months. The quality of the components is surprising. :)

tascheri, 2022-05-06:
Hello, 
Im having the same problems. I was measuring the fan ouputs with a multimeter and somehow the USB connection to the board is not working anymore. Also the fimware update is not working, no green light, nothing. All the lights ( 3,3v 5v and 12 ) are lit.

DEAth1981, 2022-05-06:
> Hello, Im having the same problems. I was measuring the fan ouputs with a multimeter and somehow the USB connection to the board is not working anymore. Also the fimware update is not working, no green light, nothing. All the lights ( 3,3v 5v and 12 ) are lit. The 3.3 volts pins are showing correct voltage

Hi, my problem was solved by replacing the STM32 on the board

tascheri, 2022-05-06:
> Hi, my problem was solved by replacing the STM32 on the board

Ouch... IDK if im capable (or even have the tools ) to do that. I gather the bootloader will have to be uploaded to the STM once installed.  Is there a tutorial somewhere to do the replacement / bootloader upload?

DEAth1981, 2022-05-06:
> Ouch... IDK if im capable (or even have the tools ) to do that. I gather the bootloader will have to be uploaded to the STM once installed. Is there a tutorial somewhere to do the replacement / [...]

https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders/GTR%20V1.0

Talha909, 2022-07-02:
> SMD resistor

did you find any solution. I have the same issue