bigtreetech/BIGTREETECH-GTR-V1.0 #63: No/bad overvoltage protection?

mukha521, 2021-04-13:
A sudden quick print head move (X and Y motors) blew up the U2 (SGM6130) power chip leaving the board not operational...
Is there proper protection from such voltage coming from the motors?

Drivers are TMC5160 which survived the accident!

Gaulishdance-fablab, 2022-12-14:
I have exactly the same question/problem as you... Did you find any answer?

mukha521, 2022-12-14:
> I have exactly the same question/problem as you... Did you find any answer?
Hello, 
Bought a new board... =(

You can try to source and solder a new SGM6130 chip hoping that only the chip was destroyed, But I had no free time for that, unfortunately.

A13xund3r, 2024-01-01:
Just happen to me after I manually moved XY very fast, all drivers (TMC2209) gone with nice holes in them, SGM6130 blown, rest of the board possibly survived, moved jumper J2 to USB position and without drivers board booted up. There is no overvoltage protection component on the schematic and I can't see any on the PCB, this is very bad.. TVS diode is not that expensive...